Joshua-Riek/ubuntu-rockchip #1026: Bug Report: Orange Pi 5 Max can't connect to WPA2 5G WiFi networks but can connect to 2.4G

DiegoBM, 2024-09-02:
### What happened?

After installing the network managert, when trying to connect to my WiFi, the 2.4G network will work fine, but the 5G network will consistently fail with the error:
"Error: Connection activation failed: Secrets were required, but not provided."

Both networks are separated, use the same encryption "WPA/WPA2-Personal", and have the same password. I've tried multiple solutions online to no avail.

According to [this link](https://wiki.archlinux.org/title/NetworkManager#Secrets_were_required,_but_not_provided), it might be related to the association taking too long, but there's no reason for it to take too long, in any case I also tried their suggestions with no success. Might not mean anything, but the board is phisically right next to the router, full signal.

### Kernel version

6.1.0-1023-rockchip

### SBC model

Orange Pi 5 Max

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

### Relevant logs

```shell
Sep 02 19:52:00 ubuntu NetworkManager[892]: <info>  [1725306720.7485] device (enP3p49s0): carrier: link connected
Sep 02 19:52:54 ubuntu NetworkManager[892]: <info>  [1725306774.8007] keyfile: deleting netplan connection: NM-c1588f29-05be-40f9-8774-6fa8bd348113
Sep 02 19:52:55 ubuntu NetworkManager[892]: <info>  [1725306775.3986] audit: op="connection-delete" uuid="c1588f29-05be-40f9-8774-6fa8bd348113" name="mywifi" pid=1724 uid=0 resul>
Sep 02 19:53:37 ubuntu NetworkManager[892]: <info>  [1725306817.5757] device (wlan0): Activation: starting connection 'mywifi' (2af33430-51c7-495b-b8e2-034937470da8)
Sep 02 19:53:37 ubuntu NetworkManager[892]: <info>  [1725306817.5759] audit: op="connection-add-activate" uuid="2af33430-51c7-495b-b8e2-034937470da8" name="mywifi" pid=1789 uid=0>
Sep 02 19:53:37 ubuntu NetworkManager[892]: <info>  [1725306817.5764] device (wlan0): state change: disconnected -> prepare (reason 'none', sys-iface-state: 'managed')
Sep 02 19:53:37 ubuntu NetworkManager[892]: <info>  [1725306817.5766] manager: NetworkManager state is now CONNECTING
Sep 02 19:53:37 ubuntu NetworkManager[892]: <info>  [1725306817.5769] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Sep 02 19:53:37 ubuntu NetworkManager[892]: <info>  [1725306817.5779] device (wlan0): Activation: (wifi) access point 'mywifi' has security, but secrets are required.
Sep 02 19:53:37 ubuntu NetworkManager[892]: <info>  [1725306817.5780] device (wlan0): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Sep 02 19:53:37 ubuntu NetworkManager[892]: <info>  [1725306817.5781] sup-iface[e27e97b7051a3d97,0,wlan0]: wps: type pbc start...
Sep 02 19:53:37 ubuntu NetworkManager[892]: <info>  [1725306817.5791] device (wlan0): state change: need-auth -> prepare (reason 'none', sys-iface-state: 'managed')
Sep 02 19:53:37 ubuntu NetworkManager[892]: <info>  [1725306817.5793] device (wlan0): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Sep 02 19:53:37 ubuntu NetworkManager[892]: <info>  [1725306817.5796] device (wlan0): Activation: (wifi) connection 'mywifi' has security, and secrets exist.  No new secrets need>
Sep 02 19:53:37 ubuntu NetworkManager[892]: <info>  [1725306817.8988] Config: added 'ssid' value 'mywifi'
Sep 02 19:53:37 ubuntu NetworkManager[892]: <info>  [1725306817.8989] Config: added 'scan_ssid' value '1'
Sep 02 19:53:37 ubuntu NetworkManager[892]: <info>  [1725306817.8989] Config: added 'bgscan' value 'simple:30:-70:86400'
Sep 02 19:53:37 ubuntu NetworkManager[892]: <info>  [1725306817.8989] Config: added 'key_mgmt' value 'WPA-PSK WPA-PSK-SHA256 FT-PSK SAE FT-SAE'
Sep 02 19:53:37 ubuntu NetworkManager[892]: <info>  [1725306817.8989] Config: added 'auth_alg' value 'OPEN'
Sep 02 19:53:37 ubuntu NetworkManager[892]: <info>  [1725306817.8990] Config: added 'psk' value '<hidden>'
Sep 02 19:53:45 ubuntu NetworkManager[892]: <info>  [1725306825.2385] device (wlan0): supplicant interface state: disconnected -> scanning
Sep 02 19:53:45 ubuntu NetworkManager[892]: <info>  [1725306825.2386] device (p2p-dev-wlan0): supplicant management interface state: disconnected -> scanning
Sep 02 19:53:52 ubuntu NetworkManager[892]: <info>  [1725306832.0791] device (wlan0): supplicant interface state: scanning -> associating
Sep 02 19:53:52 ubuntu NetworkManager[892]: <info>  [1725306832.0792] device (p2p-dev-wlan0): supplicant management interface state: scanning -> associating
Sep 02 19:54:02 ubuntu NetworkManager[892]: <info>  [1725306842.6640] device (wlan0): supplicant interface state: associating -> disconnected
Sep 02 19:54:02 ubuntu NetworkManager[892]: <info>  [1725306842.6641] device (p2p-dev-wlan0): supplicant management interface state: associating -> disconnected
Sep 02 19:54:02 ubuntu NetworkManager[892]: <info>  [1725306842.9258] device (wlan0): supplicant interface state: disconnected -> scanning
Sep 02 19:54:02 ubuntu NetworkManager[892]: <info>  [1725306842.9259] device (p2p-dev-wlan0): supplicant management interface state: disconnected -> scanning
Sep 02 19:54:03 ubuntu NetworkManager[892]: <warn>  [1725306843.2866] device (wlan0): Activation: (wifi) association took too long
Sep 02 19:54:03 ubuntu NetworkManager[892]: <info>  [1725306843.2866] device (wlan0): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Sep 02 19:54:03 ubuntu NetworkManager[892]: <info>  [1725306843.2869] sup-iface[e27e97b7051a3d97,0,wlan0]: wps: type pbc start...
Sep 02 19:54:03 ubuntu NetworkManager[892]: <warn>  [1725306843.2871] device (wlan0): Activation: (wifi) asking for new secrets
Sep 02 19:54:03 ubuntu NetworkManager[892]: <warn>  [1725306843.2873] device (wlan0): no secrets: No agents were available for this request.
Sep 02 19:54:03 ubuntu NetworkManager[892]: <info>  [1725306843.2873] device (wlan0): state change: need-auth -> failed (reason 'no-secrets', sys-iface-state: 'managed')
Sep 02 19:54:03 ubuntu NetworkManager[892]: <info>  [1725306843.2876] manager: NetworkManager state is now DISCONNECTED
Sep 02 19:54:03 ubuntu NetworkManager[892]: <warn>  [1725306843.2879] device (wlan0): Activation: failed for connection 'mywifi'
Sep 02 19:54:03 ubuntu NetworkManager[892]: <info>  [1725306843.2881] device (wlan0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
Sep 02 19:54:09 ubuntu NetworkManager[892]: <info>  [1725306849.6762] device (wlan0): supplicant interface state: scanning -> disconnected
Sep 02 19:54:09 ubuntu NetworkManager[892]: <info>  [1725306849.6763] device (p2p-dev-wlan0): supplicant management interface state: scanning -> disconnected
```

DiegoBM, 2024-09-02:
Ubuntu Desktop can connect to the 5G network (not sure how), so if it's using the same  drivers and all the same, I presume that it must be something that I'm doing wrong

DiegoBM, 2024-09-03:
Actually the exact same operation in Orane Pi's ubuntu server distro just works, so maybe I'm not doing anything wrong

Joshua-Riek, 2024-09-03:
Ubuntu 24.04 uses netplan, you should not need to install a network manager?

DiegoBM, 2024-09-03:
Apologies, this is my first time connecting to WiFi through command line, I was in fact following the only ubuntu guide I could find on  [configuring WiFi connections](https://ubuntu.com/core/docs/networkmanager/configure-wifi-connections), so I assumed it was the way to go. Where could I find more details about netplan? Is nmcli not supposed to work then?

aggiedude, 2024-09-06:
I seem to be having the same issue with my Orange Pi 5 Max, have switched to using the ethernet port for now.

ewaldc, 2024-09-23:
@Joshua-Riek, in Ubuntu 24.04 NetworkManager actually uses a [YAML/netplan ](https://ubuntu.com/core/docs/networkmanager/networkmanager-and-netplan) based back-end. So even in the desktop version, the network configuration will be saved/stored in `/etc/netplan` directory as you modify the config using `nmtui` and/or GUI.  @DiegoBM, take a look at that folder, if all went well, you should see your Wlan config file there.  It should be similar to that in Ubuntu server.

racky1, 2024-10-28:
I have the same issue on Ubuntu 24.04 with wifi.  It doesn't appear to connect in anything but 2.4ghz.  I have wifi 6e and other bad versions of Max OS use the 6GHz but are bad versions of OS.  Any idea?  I don't have any errors that I see.

closed 2025-02-10 by github-actions[bot]

mammuthus, 2025-06-05:
I have the same issue on my **Orange Pi 5B** with **Ubuntu 24.04.2 LTS**
`Linux pi 6.1.0-1025-rockchip #25-Ubuntu SMP Mon Aug 26 23:01:14 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux`

I use nmcli, and there are no issues connecting to 2.4 GHz. But when I try to connect to 5 GHz (same router), I get an error "Error: Connection activation failed: Secrets were required, but not provided." (Sure, I provide the correct password).

The caveat could be that my access point uses mixed encryption (WPA2/WPA3), but I couldn’t test it with just WPA2 (or WPA3) because it’s my ISP’s router and I have no internal access. 

@DiegoBM Does your router use mixed encryption too? And did you manage to solve the issue?