Joshua-Riek/ubuntu-rockchip #1028: Bug Report: Orange PI 5 Plus constantly crushinhg on 6.1.0-1023-Rockchip Ubuntu 22.04 and 24.04

dmadma1, 2024-09-05:
### What happened?

Wanted to use last desktop Joshua images on my orange but no chance - after boot, crashing in few minutes with no reason or crashing immediatelly under some load: like staring youtube in browser.
Noticed led is blinking more fast before crush but it is not always related. Power supply recently upgraded from 3A to 4A model but crushes persist.

I had no such issues on Last Armbian images (Armbian 24.8.1 Noble Gnome, MESA / VPU, Kernel: 6.1.75, Size: 1.5 GB, Release date: Aug 26, 2024)  with 3A power supply but here I had max resolution 1920x1080 and it is not enough for me - I need 2560 or 3440x1400p

### Kernel version

6.1.0-1023-Rockchip

### SBC model

Orange PI 5 Plus

### What operating system are you seeing this problem on?

Ubuntu 22.04 LTS (Jammy Jellyfish), Ubuntu 24.04 LTS (Noble Nombat)

dmadma1, 2024-09-05:
[   18.295475] vdd_2v0_pldo_s3: could not add device link regulator.31: -ENOENT
[   18.295983] avcc_1v8_s0: 1800 mV, enabled
[   18.296226] rk806-regulator rk806-regulator.2.auto: Looking up vcc11-supply from device tree
[   18.296237] vcc_1v8_s0: supplied by vdd_2v0_pldo_s3
[   18.296242] vdd_2v0_pldo_s3: could not add device link regulator.32: -ENOENT
[   18.296753] vcc_1v8_s0: 1800 mV, enabled
[   18.296977] rk806-regulator rk806-regulator.2.auto: Looking up vcc11-supply from device tree
[   18.296988] avdd_1v2_s0: supplied by vdd_2v0_pldo_s3
[   18.296993] vdd_2v0_pldo_s3: could not add device link regulator.33: -ENOENT
[   18.297436] avdd_1v2_s0: 1200 mV, enabled
[   18.297662] rk806-regulator rk806-regulator.2.auto: Looking up vcc12-supply from device tree
[   18.297672] vcc_3v3_s0: supplied by vcc5v0_sys
[   18.297678] vcc5v0_sys: could not add device link regulator.34: -ENOENT
[   18.298047] vcc_3v3_s0: 3300 mV, enabled
[   18.298272] rk806-regulator rk806-regulator.2.auto: Looking up vcc12-supply from device tree
[   18.298280] vccio_sd_s0: supplied by vcc5v0_sys
[   18.298285] vcc5v0_sys: could not add device link regulator.35: -ENOENT
[   18.298654] vccio_sd_s0: 1800 <--> 3300 mV at 3300 mV, enabled
[   18.298889] rk806-regulator rk806-regulator.2.auto: Looking up vcca-supply from device tree
[   18.298898] pldo6_s3: supplied by vcc5v0_sys
[   18.298903] vcc5v0_sys: could not add device link regulator.36: -ENOENT
[   18.299080] pldo6_s3: 1800 mV, enabled
[   18.299340] rk806 spi2.0: no sleep-setting state
[   18.299363] rk806 spi2.0: no reset-setting pinctrl state
[   18.299373] rk806 spi2.0: no dvs-setting pinctrl state
[   18.301395] rockchip-spi feb20000.spi: register misc device rkspi-dev2
[   18.301402] rockchip-spi feb20000.spi: probed, poll=0, rsd=0, cs-inactive=0, ready=0
[   18.302893] spi-nor spi5.0: XM25QU128C (16384 Kbytes) read_data x4
[   18.302932] 1 fixed-partitions partitions found on MTD device sfc_nor
[   18.302939] Creating 1 MTD partitions on "sfc_nor":
[   18.302943] 0x000000000000-0x000001000000 : "loader"
[   18.305891] usbcore: registered new interface driver rndis_wlan
[   18.305914] usbcore: registered new interface driver asix
[   18.305930] usbcore: registered new interface driver cdc_ether
[   18.305947] usbcore: registered new interface driver rndis_host
[   18.305972] usbcore: registered new interface driver cdc_ncm
[   18.306135] VFIO - User Level meta-driver version: 0.3
[   18.315375] phy phy-fd5d4000.syscon:usb2-phy@4000.3: illegal mode
[   18.315386] xhci-hcd xhci-hcd.3.auto: xHCI Host Controller
[   18.315399] xhci-hcd xhci-hcd.3.auto: new USB bus registered, assigned bus number 1
[   18.315493] xhci-hcd xhci-hcd.3.auto: hcc params 0x0220fe64 hci version 0x110 quirks 0x0000800002010010
[   18.315521] xhci-hcd xhci-hcd.3.auto: irq 75, io mem 0xfc400000
[   18.315606] xhci-hcd xhci-hcd.3.auto: xHCI Host Controller
[   18.315614] xhci-hcd xhci-hcd.3.auto: new USB bus registered, assigned bus number 2
[   18.315622] xhci-hcd xhci-hcd.3.auto: Host supports USB 3.0 SuperSpeed
[   18.315721] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.01
[   18.315729] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[   18.315735] usb usb1: Product: xHCI Host Controller
[   18.315739] usb usb1: Manufacturer: Linux 6.1.0-1023-rockchip xhci-hcd
[   18.315744] usb usb1: SerialNumber: xhci-hcd.3.auto
[   18.316077] hub 1-0:1.0: USB hub found
[   18.316098] hub 1-0:1.0: 1 port detected
[   18.316309] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[   18.316375] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.01
[   18.316382] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[   18.316388] usb usb2: Product: xHCI Host Controller
[   18.316392] usb usb2: Manufacturer: Linux 6.1.0-1023-rockchip xhci-hcd
[   18.316397] usb usb2: SerialNumber: xhci-hcd.3.auto
[   18.316663] hub 2-0:1.0: USB hub found
[   18.316684] hub 2-0:1.0: 1 port detected
[   18.316718] ehci-platform fc800000.usb: EHCI Host Controller
[   18.316750] ohci-platform fc840000.usb: Generic Platform OHCI controller
[   18.316852] ohci-platform fc840000.usb: new USB bus registered, assigned bus number 3
[   18.316857] ehci-platform fc800000.usb: new USB bus registered, assigned bus number 4
[   18.316902] usbcore: registered new interface driver cdc_acm
[   18.316908] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[   18.316927] ehci-platform fc800000.usb: irq 76, io mem 0xfc800000
[   18.317023] ohci-platform fc840000.usb: irq 78, io mem 0xfc840000
[   18.317052] usbcore: registered new interface driver uas
[   18.317088] usbcore: registered new interface driver usb-storage
[... 812 lines cut ...]
[   23.449453] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[   23.476205] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[   23.506068] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[   23.532736] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[   23.559394] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[   23.586075] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[   23.612749] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[   23.639423] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[   23.666075] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[   23.692745] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[... cut ...]
[   28.743977] systemd-journald[407]: /var/log/journal/f83f802b60384d72b24d9c75443705c8/user-1000.journal: Journal file uses a different sequence number ID, rotating.
[   29.673087] r8169 0003:31:00.0 enP3p49s0: Link is Up - 1Gbps/Full - flow control rx/tx
[   29.673115] IPv6: ADDRCONF(NETDEV_CHANGE): enP3p49s0: link becomes ready
[   32.599600] mali fb000000.gpu: Loading Mali firmware 0x1010000
[   32.601007] mali fb000000.gpu: Mali firmware git_sha: ee476db42870778306fa8d559a605a73f13e455c
[   33.103281] platform mtd_vendor_storage: deferred probe pending
root@orpi5:/home/raf#

dmadma1, 2024-09-05:
What log should I chect to verify I have no same issue as here?:
https://github.com/Joshua-Riek/ubuntu-rockchip/issues/502

I erased spi by dd and tried image on sd card and nvme ssd - same  crash problems

dmadma1, 2024-09-05:
Can somebody recommend me some older image with GPU support which can be more stable  on Orange pi 5 plus ?

Joshua-Riek, 2024-09-05:
If you can provide useful crash logs I can see what's going on. But the Orange Pi 5 Plus has been the device I use for most development, so I'm very confident in it's stability.

SimplyCorbett, 2024-09-07:
I have had no crashing on the orange pi 5 plus, but I also use several of them for servers with no GUI or monitor connected.

In the ops case it may be useful for them to see what specific kernel upgrade caused the issue (original working kernel -> problem kernel) to help narrow it down, along with a better error log.

dmadma1, 2024-09-08:
Become not actual for me

closed 2024-09-08 by dmadma1