Joshua-Riek/ubuntu-rockchip #1034: Bug Report: Rock5C-Lite

Qengineering, 2024-09-07:
### What happened?

The support for the Rock5C-Lite still fails:

After the initialization steps Ubuntu 22.04 [release v2.3.2 2024-09-04](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v2.3.1/ubuntu-22.04-preinstalled-desktop-arm64-rock-5c.img.xz) reboots and the Rock5C-Lite errors with:
fail to start casper-m5check Verifiy Live ISO checksums.

The Ubuntu 24.04 [release v2.3.2 2024-09-04](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v2.3.1/ubuntu-24.04-preinstalled-desktop-arm64-rock-5c.img.xz) shows in a glimpse the Ubuntu 24.04 screen, jumps to command line messages and stocks with only a blinking cursor in the top. No initialization whatsoever.

### Kernel version

unable to fetch

### SBC model

Rock5C-Lite

### What operating system are you seeing this problem on?

Ubuntu 22.04 LTS (Jammy Jellyfish), Ubuntu 24.04 LTS (Noble Nombat)

ModFol, 2024-09-08:
I just tried to boot a rock 5c lite from an nvme drive. I could not get it to work. I have attached a copy of the uart console output in case that is helpful.
[5c-lite.txt](https://github.com/user-attachments/files/16922013/5c-lite.txt)

kisturyo, 2024-09-13:
It needs to flash a patched U-Boot to unlock the GPU Core if you want to enter desktop environment.

krishna4104, 2024-12-20:
> It needs to flash a patched U-Boot to unlock the GPU Core if you want to enter desktop environment.
@kisturyo Did u find a way around or anything that can solve this

kisturyo, 2025-01-04:
> > It needs to flash a patched U-Boot to unlock the GPU Core if you want to enter desktop environment.
> > @kisturyo Did u find a way around or anything that can solve this

Armbian's u-boot has patched.You can use it.

Qengineering, 2025-01-04:
@kisturyo,
Please, enlighten your pearls of wisdom.  
Most readers aren't as senior in the Debian / Linux world as you are. 
They need more guidance, more a step to step approach.

closed 2025-03-10 by github-actions[bot]