Joshua-Riek/ubuntu-rockchip #1043: Bug Report: Poor hevc encoding performance

MentosGuy, 2024-09-14:
### What happened?

I have tested Gstreamer pipelines, both H264 decoded feeds via RTMP and HDMIrx inputs with a target bitrate of 8 Mbit (resolution in both cases 1080p60), as soon as detailed objects such as forests with leaves appear, the picture becomes totally blurred despite the high bitrate. I use mpph265enc as an encoder. Is this normal because the VPU is designed for speed rather than detail accuracy, or can better performance be achieved with kernel updates for mpp and rga or by replacing the Mali GPU driver? 

### Kernel version

rk-6.1-rkr1 (Pulled in January 2024)

### SBC model

Orange Pi 5 Plus

### What operating system are you seeing this problem on?

Ubuntu 22.04 LTS (Jammy Jellyfish)

defencedog, 2024-09-14:
You can follow [this procedure](https://github.com/defencedog/orangepi3b_v2.1/tree/main/files_tools/vendor_Kernel6.1/Overclocked_dtb) here to overclock Video Decoder (not GPU)... Method is explained in detail & can be adopted for OPi5

Bluesmanuk, 2024-09-16:
I was intrigued to see what effect this change may have upon general video performance on the Orange Pi 5.

But if my interpretation of what is required is correct, then under the Video Decoder Overclocking section of the linked article, there is no rc.local to add the necessary line of information to.

It may well be more complicated than that and I am reading things incorrectly and as it is merely a mild interest in the topic, no response is required. Just an observation based upon what I interpreted.

defencedog, 2024-09-16:
@Bluesmanuk `rc.local` is as old as Linux itself. It's a way to run scripts at startup. Visit [this](https://www.cyberciti.biz/faq/how-to-enable-rc-local-shell-script-on-systemd-while-booting-linux-system/) or just Google it

Joshua-Riek, 2024-09-17:
Closing as invalid. Please provide exact reproduceable steps and system logs. I've not encountered problems with encoding.

closed 2024-09-17 by Joshua-Riek

Bluesmanuk, 2024-09-19:
@defencedog

Thanks for the link.

I applied the code but found that video decoder performance was much worse.

But worth the test to find out.

defencedog, 2024-09-25:
@Bluesmanuk what do you mean much verse? After overclocking video decoding should be superior

Bluesmanuk, 2024-09-25:
@defencedog

There was major stuttering and frame dropping all over the place.