Joshua-Riek/ubuntu-rockchip #1044: When will the image of NanoPi M6 be released?

LeeXujie, 2024-09-14:


Joshua-Riek, 2024-09-14:
This is one of the more interesting boards because of the built in display, I need to get my hands on the hardware then work on drivers.

LeeXujie, 2024-09-18:
> This is one of the more interesting boards because of the built in display, I need to get my hands on the hardware then work on drivers.

Yes, their development board hardware is indeed doing well, but their official system seems to have poor GPU support. When I run the VTK program on Ubuntu 22.04, the interactive interface is very laggy. Can you give some suggestions to solve this problem?

[image]

lukaszsobala, 2024-10-24:
@LeeXujie Just add a ppa with panfork mesa drivers, for example this one, and install drivers that actually work for rk3588.

https://launchpad.net/~jjriek/+archive/ubuntu/panfork-mesa/

closed 2025-02-10 by github-actions[bot]