Joshua-Riek/ubuntu-rockchip #1045: Bug Report: NFS not working

Mqxx, 2024-09-14:
### What happened?

After connectiong a network drive using nfs it should work. Insed the terminal just hangs up and nothing happens when trying to connect the drive.

How I connected the drive before (on a Raspberry Pi 5):
Follow [this](https://linuxize.com/post/how-to-mount-an-nfs-share-in-linux/) tutorial. This should also work on the Orange Pi 5 Pro but it does not, or did I something wrong? Are there any alternative solutions I can try?

### Kernel version

5.10.0-1012-rockchip

### SBC model

Orange Pi 5 Pro

### What operating system are you seeing this problem on?

Ubuntu 22.04 LTS (Jammy Jellyfish)

ewaldc, 2024-09-23:
@Mqxx, I have moved on to Ubuntu 24.04 now but have run NFS for almost a year on 22.04 without any issues, including RPI clients (two of my Orange PI 5+ are NAS servers).  Hence, it's more likely some configuration issue.

On the server: what does the command `exportfs` say ?   Can you provide the contents of `/etc/exports`.  Did you run `exports -a` ? Status of NFS server (`systemctl status nfs-server`)?
On the client: what does `showmount -e <your NFS server IP address>` give?  What is your mount command?
Some additional things to check: do the user/group id's match between client/server, is there a router/firewall between client and server ? Is DNS lookup working (both forward and reverse) ?  Are you using PAM and/or Kerberos ? If so, is `libpam-krb5` installed (in Jammy there was a missing package depency ...)?  Are the outputs of `rpcinfo -p localhost` on both client and server matching (versions, ports)?
I would recommend to try using NFS version 3 first, e.g. mount using `noatime,rw,soft,bg,nolock,intr,async,nfsvers=3,rsize=65536,wsize=65536`

A hang generally means some issue with portmapper or authentication.

Mqxx, 2024-09-28:
I was able to fix it using:
```
XXX.XXX.XXX.XXX:/path/from/nas /mnt/path nfs vers=3,timeo=900,retrans=3 0 0
```

closed 2024-09-28 by Mqxx