Joshua-Riek/ubuntu-rockchip #1046: Feature Request: Enable KSMBD in kernel configuration

shirok1, 2024-09-15:
### Feature Description

KSMBD is a kernel-space SMB (aka the protocol to "share" folders to Windows) server, upstreamed into Linux kernel. Its user-space tool is already in Ubuntu repo, 2204 and 2404.

I think enabling it in kernel configuration can benefit those wanted to use their SBC as a NAS!

My alternatives:

- Use Samba instead, which is user-space, and may be less performant
- Build as a kmod manually
- Try to install as DKMS (I've tried this on the deprecated version of 2204 before, it needs to be reinstalled after kernel upgrade, maybe I miss-configured it)

shirok1, 2024-09-15:
FYI: https://github.com/cifsd-team/ksmbd.git

Joshua-Riek, 2024-09-21:
Can you please send a pull request to the kernel?

shirok1, 2024-09-21:
> Can you please send a pull request to the kernel?

I think we can just change...

https://github.com/Joshua-Riek/linux-rockchip/blob/2a5e23f029ac47829348b87dd56409698fd7e3bf/debian.rockchip/config/config.common.ubuntu#L8700

to sth. like `CONFIG_SMB_SERVER=m`. Sorry I can't test it out right now.

closed 2025-02-10 by github-actions[bot]