Joshua-Riek/ubuntu-rockchip #1050: Missing?  No dtoverlays under /boot, actually no firmware directory under /boot as well?

Jibun-no-Kage, 2024-09-18:
### What happened?

Missing?  No dtoverlays under /boot, actually no firmware directory under /boot as well?  No dtoverlys at all in 22.04 ubuntu image for IndieDroid Nova?

### Kernel version

5.10.0-1012-rockchip

### SBC model

IndieDroid NOVA

### What operating system are you seeing this problem on?

Ubuntu 22.04 LTS (Jammy Jellyfish)

### Relevant logs

```shell
None... if not dtoverlays present, what logs?
```

Joshua-Riek, 2024-09-18:
There was a lot of changes to the filesystem structure. Please read the wiki, this is expected.

closed 2024-09-18 by Joshua-Riek

Jibun-no-Kage, 2024-09-18:
Which Wiki?  This one?  https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-22.04-LTS.  Unfortunately, I do not know if I need to 'change' the path... I don't even know if the files are even in the image.  How do you validate the files are there if they are?

I am familiar with u-boot-update, but every SBC I have used thus far had the dtbo files in the same basic location, i.e. /boot/ or below that file system.  But not the images you generated?  Your wiki states if I need to change the path?  Do I need to change the path?  I don't know what path is the correct one, or if I need to change it.

I find no dbto files on the current image I have, i.e. ubuntu desktop 22.04...
```
\# uname -r 
5.10.0-1012-rockchip
\# cat /etc/os-release
PRETTY_NAME="Ubuntu 22.04.5 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.5 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy

\# find / | grep dbto
find: ‘/run/user/1000/doc’: Permission denied
find: ‘/run/user/1000/gvfs’: Permission denied
```
With respect, I need a point of reference, a place to start, before I know if I need to change anything.  I don't see anywhere where you explain what the default path is or how to determine it.  How to find files or know if they exist?  Do I load a package?  No where in the wiki is this explained.

More over I know I need to use i2c bus 8, per the pinout documentation for the IndieDroid Nova.  So I would think the right overlay would reference i2c8 or some variant of that for the file name I need.  But searching for i2c8, I find a lot of files, but none are extension .dbto, so again, unable to determine next steps.

Some help or explanation would be appreciated.

Joshua-Riek, 2024-09-18:
I'm trying to take a break.. Read the wiki again.. Also use the correct extension, it's `.dtbo` not `.dbto`.

Jibun-no-Kage, 2024-09-18:
OK... sorry about a typo.  Once I understand this... I would have no problem updating the wiki... is that would help you out?