Joshua-Riek/ubuntu-rockchip #1051: Debugging the brcmfmac driver

jamstec0, 2024-09-18:
Hello. 
In order to investigate the cause of the Wi-Fi error on Ubuntu 24.04, I used the kernel config from linux-rockchip and enabled debugging to rebuild the `brcmfmac` and `brcmutil` modules, but a memory access error occurred, and I was unable to perform debugging.
 The error log has been deleted, so I will upload it later. Could the build settings be different from those of the drivers originally in the kernel?

Joshua-Riek, 2024-09-18:
What board..?

ewaldc, 2024-09-23:
@jamstec0, the kernel is based on [linux-rockchip](https://github.com/Joshua-Riek/linux-rockchip), `noble` branch.  Native compilation is out of the box, but I had to make several changes for cross compilation (gcc-14.2.0) and to avoid every module (re)build causes a whole recompile and to support custom kernels with a tag e.g. `6.1.0-1025+debug-rockchip`.

jamstec0, 2024-11-27:
This bug seems to be caused by the current wpa_supplicant and is also occurring on raspberry pi devices and other devices.
As a solution, adding `brcmfmac.feature_disable=0x82000` to the boot parameter will prevent this problem.
Hopefully others will not have to spend 4 months looking for the cause.
@Joshua-Riek Sorry for the inconvenience.