Joshua-Riek/ubuntu-rockchip #1057: Bug Report: radxa ROCK 5B+ no sound input device

jhcho-hanvision, 2024-09-23:
### What happened?

With ROCK 5B+, there is no audio input device(hdmi in).
For orange pi 5 plus and ROCK 5B, it's ok.
[image]

Here is the output with arecord -l
```
root@5Bplus:~# arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: rockchipes8316 [rockchip-es8316], device 0: dailink-multicodecs ES8316 HiFi-0 [dailink-multicodecs ES8316 HiFi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: rockchiphdmiin [rockchip-hdmiin], device 0: fddf8000.i2s-dummy_codec dummy_codec-0 [fddf8000.i2s-dummy_codec dummy_codec-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
root@5Bplus:~#

```

### Kernel version

6.1.0-1025-rockchip

### SBC model

Radxa ROCK 5B Plus

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

### Relevant logs

```shell
Here are some logs copied from /var/log/dmesg

[   22.123618] kernel: clk_i2s7_8ch_rx_frac p_rate(131071999), rate(0), maybe invalid frequency setting!
[   22.123652] kernel: clk_i2s7_8ch_rx_frac p_rate(131071999), rate(0), maybe invalid frequency setting!
[   22.123669] kernel: rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   22.123701] kernel:  fddf8000.i2s-dummy_codec: ASoC: error at __soc_pcm_hw_params on fddf8000.i2s-dummy_codec: -22
[   22.123729] kernel: clk_i2s7_8ch_rx_frac p_rate(131071999), rate(0), maybe invalid frequency setting!
[   22.123745] kernel: clk_i2s7_8ch_rx_frac p_rate(131071999), rate(0), maybe invalid frequency setting!
[   22.123759] kernel: rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   22.123765] kernel:  fddf8000.i2s-dummy_codec: ASoC: error at __soc_pcm_hw_params on fddf8000.i2s-dummy_codec: -22
[   22.123787] kernel: clk_i2s7_8ch_rx_frac p_rate(131071999), rate(0), maybe invalid frequency setting!
[   22.123800] kernel: clk_i2s7_8ch_rx_frac p_rate(131071999), rate(0), maybe invalid frequency setting!
[   22.123812] kernel: rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   22.123818] kernel:  fddf8000.i2s-dummy_codec: ASoC: error at __soc_pcm_hw_params on fddf8000.i2s-dummy_codec: -22
[   22.123835] kernel: clk_i2s7_8ch_rx_frac p_rate(131071999), rate(0), maybe invalid frequency setting!
[   22.123848] kernel: clk_i2s7_8ch_rx_frac p_rate(131071999), rate(0), maybe invalid frequency setting!
[   22.123861] kernel: rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   22.123866] kernel:  fddf8000.i2s-dummy_codec: ASoC: error at __soc_pcm_hw_params on fddf8000.i2s-dummy_codec: -22
......
```

hanvinke, 2024-10-01:
@jhcho-hanvision 

To have the input device detected like this: 

[image]

..you need to enable the capture device in alsamixer. Click on F5 and choose ¨Capture¨. Then click on the spacebar. 

When enabled the ¨--------" will change like this: 

[image]

jhcho-hanvision, 2024-10-01:
Thank you for your reply.
Here is the snapshot from radxa 5B+ 
[image]

I can enable Headset Microphone as your explanation.

But, what i want to have is hdmi input device.
Here is the same snapshot with orange pi 5+ with hdmi input device.

jhcho-hanvision, 2024-10-04:
I compared the dts files for rock-5b and rock-5b-plus from these links.
[https://github.com/radxa/kernel/blob/linux-6.1-stan-rkr1/arch/arm64/boot/dts/rockchip/rk3588-rock-5b-plus.dts](url)
[https://github.com/radxa/kernel/blob/linux-6.1-stan-rkr1/arch/arm64/boot/dts/rockchip/rk3588-rock-5b.dts](url)
[https://github.com/armbian/linux-rockchip/blob/rk-6.1-rkr3/arch/arm64/boot/dts/rockchip/rk3588-rock-5b.dts](url)
[https://github.com/Joshua-Riek/linux-rockchip/blob/noble/arch/arm64/boot/dts/rockchip/rk3588-rock-5b.dts](url)
[https://github.com/Joshua-Riek/linux-rockchip/blob/noble/arch/arm64/boot/dts/rockchip/rk3588-rock-5b-plus.dts](url)
no dts file for rock-5b-plus at armbian.

I don't know well about linux kernel, but i guess rock-5b-plus.dts file is the reason for no hdmi-in audio.
Hope this search can fix this problem.
Thanks.

vtv3000, 2024-10-12:
I have the same problem ! No hdmi input audio in system

vtv3000, 2024-10-12:
And I don't have analog audio input too

jhcho-hanvision, 2024-10-14:
> And I don't have analog audio input too

You can try your audio jack plugged.

vtv3000, 2024-10-14:
Ok I will try to plugin audio jack 

But hdmi cable was plug in and no audio . 
In alsamixer I can see hdmi input but not have control . Gstreamer cannot have sound from hdmi 

May be need EDDID file ?

jhcho-hanvision, 2024-10-14:
> https://github.com/Joshua-Riek/linux-rockchip/blob/noble/arch/arm64/boot/dts/rockchip/rk3588-rock-5b-plus.dts

I have the same problem with hdmi input.
I am waiting for Joshua-Riek's update for hdmi audio input problem.

vtv3000, 2024-10-14:
> > https://github.com/Joshua-Riek/linux-rockchip/blob/noble/arch/arm64/boot/dts/rockchip/rk3588-rock-5b-plus.dts
> 
> 
> 
> I have the same problem with hdmi input.
> 
> I am waiting for Joshua-Riek's update for hdmi audio input problem.

I compare .dts files and see some different

[image]

vtv3000, 2024-10-23:
Today fresh install Ubuntu and I see audio input from analog input

But audio from hdmi input not work ..

closed 2025-02-10 by github-actions[bot]

moo-the-cow, 2025-11-14:
I encountered this issue too, the problem was that the overlay configuration was using the DUMMY codec.
I have fixed that and compiled a tdbo that you can place in your overlays directory and reference in your u-boot config (see Joshuas documentation).

binary:
https://github.com/moo-the-cow/streaming-tools/raw/refs/heads/main/misc/hdmiin-fix.dtbo
source code:
https://github.com/moo-the-cow/streaming-tools/raw/refs/heads/main/misc/hdmiin-fix.dts

enjoy hdmi-in audio actually having sound