Joshua-Riek/ubuntu-rockchip #1060: Bug Report: EMMC CQE Recovery resulting in read-only filesystem

tomharvey, 2024-09-26:
### What happened?

I was doing some intensive disk activity on my EMMC device (no other disks are connected) when it flipped to read-only mode.

Kernel logging is attached 

This seems similar to issues which resulted in a reconfiguration of the EMMC in the official kernel here https://github.com/radxa/kernel/pull/127/files and issues reported in Armbian here https://forum.armbian.com/topic/18745-emmc-errors/

### Kernel version

6.1.0-1025-rockchip

### SBC model

Radxa Rock 5B

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

### Relevant logs

```shell
2024-09-26T11:32:35.548496+00:00 ubuntu kernel: mmc0: running CQE recovery
2024-09-26T11:32:36.067121+00:00 ubuntu kernel: mmc0: cqhci: Failed to halt
2024-09-26T11:32:36.067167+00:00 ubuntu kernel: ------------[ cut here ]------------
2024-09-26T11:32:36.067172+00:00 ubuntu kernel: mmc0: cqhci: spurious TCN for tag 9
2024-09-26T11:32:36.067182+00:00 ubuntu kernel: WARNING: CPU: 0 PID: 6139 at drivers/mmc/host/cqhci-core.c:791 cqhci_irq+0x260/0x2e8
2024-09-26T11:32:36.067184+00:00 ubuntu kernel: Modules linked in: veth xt_conntrack nft_chain_nat xt_MASQUERADE nf_nat bridge stp llc nf_conntrack_netlink nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xt_addrtype nft_compat nf_tables overlay btusb btmtk qrtr bnep binfmt_misc pwm_fan hci_uart rtw89_8852be btqca rtw89_8852b rtw89_pci rtw89_core rtk_btusb panfrost btrtl btbcm btintel drm_shmem_helper gpu_sched input_leds joydev uio_pdrv_genirq uio squashfs dm_multipath fuse dm_mod nfnetlink ip_tables ipv6 raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx raid1 raid0 multipath linear md_mod r8169 [last unloaded: btmtk]
2024-09-26T11:32:36.067186+00:00 ubuntu kernel: CPU: 0 PID: 6139 Comm: kworker/0:1H Not tainted 6.1.0-1025-rockchip #25-Ubuntu
2024-09-26T11:32:36.067188+00:00 ubuntu kernel: Hardware name: Radxa ROCK 5B (DT)
2024-09-26T11:32:36.067189+00:00 ubuntu kernel: Workqueue: kblockd blk_mq_run_work_fn
2024-09-26T11:32:36.067192+00:00 ubuntu kernel: pstate: 604000c9 (nZCv daIF +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
2024-09-26T11:32:36.067195+00:00 ubuntu kernel: pc : cqhci_irq+0x260/0x2e8
2024-09-26T11:32:36.067197+00:00 ubuntu kernel: lr : cqhci_irq+0x260/0x2e8
2024-09-26T11:32:36.067199+00:00 ubuntu kernel: sp : ffff800008003db0
2024-09-26T11:32:36.067200+00:00 ubuntu kernel: x29: ffff800008003dc0 x28: 0000000003ff003f x27: 000000000270003e
2024-09-26T11:32:36.067202+00:00 ubuntu kernel: x26: 00000000010f0001 x25: ffff0001003a9698 x24: ffff80000b298eef
2024-09-26T11:32:36.067203+00:00 ubuntu kernel: x23: ffff80000b2e6160 x22: 0000000000000009 x21: 0000000000000002
2024-09-26T11:32:36.067205+00:00 ubuntu kernel: x20: ffff0001003a9680 x19: ffff000100245000 x18: 0000000000000000
2024-09-26T11:32:36.067206+00:00 ubuntu kernel: x17: 0000000000000000 x16: 0000000000000000 x15: 0000000000000000
2024-09-26T11:32:36.067210+00:00 ubuntu kernel: x14: 0000000000000000 x13: 392067617420726f x12: 66204e4354207375
2024-09-26T11:32:36.067213+00:00 ubuntu kernel: x11: 6f6972757073203a x10: 6963687163203a30 x9 : ffff8000080ea040
2024-09-26T11:32:36.067214+00:00 ubuntu kernel: x8 : 43542073756f6972 x7 : 757073203a696368 x6 : 7163203a30636d6d
2024-09-26T11:32:36.067216+00:00 ubuntu kernel: x5 : ffff0004fdda9a90 x4 : 0000000000000000 x3 : 0000000000000027
2024-09-26T11:32:36.067218+00:00 ubuntu kernel: x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffff000104cacec0
2024-09-26T11:32:36.067220+00:00 ubuntu kernel: Call trace:
2024-09-26T11:32:36.067222+00:00 ubuntu kernel:  cqhci_irq+0x260/0x2e8
2024-09-26T11:32:36.067224+00:00 ubuntu kernel:  dwcmshc_cqe_irq_handler+0x5c/0x90
2024-09-26T11:32:36.067226+00:00 ubuntu kernel:  sdhci_irq+0x260/0x3d0
2024-09-26T11:32:36.067227+00:00 ubuntu kernel:  __handle_irq_event_percpu+0xbc/0x1c0
2024-09-26T11:32:36.067229+00:00 ubuntu kernel:  handle_irq_event_percpu+0x20/0x54
2024-09-26T11:32:36.067231+00:00 ubuntu kernel:  handle_irq_event+0x50/0x94
2024-09-26T11:32:36.067233+00:00 ubuntu kernel:  handle_fasteoi_irq+0xa4/0x124
2024-09-26T11:32:36.067234+00:00 ubuntu kernel:  handle_irq_desc+0x28/0x40
2024-09-26T11:32:36.067236+00:00 ubuntu kernel:  generic_handle_domain_irq+0x20/0x2c
2024-09-26T11:32:36.067239+00:00 ubuntu kernel:  __gic_handle_irq_from_irqson.isra.0+0x140/0x1c4
2024-09-26T11:32:36.067240+00:00 ubuntu kernel:  gic_handle_irq+0x88/0x90
2024-09-26T11:32:36.067242+00:00 ubuntu kernel:  call_on_irq_stack+0x24/0x4c
2024-09-26T11:32:36.067244+00:00 ubuntu kernel:  do_interrupt_handler+0x64/0xa8
2024-09-26T11:32:36.067245+00:00 ubuntu kernel:  el1_interrupt+0x90/0xa0
2024-09-26T11:32:36.067247+00:00 ubuntu kernel:  el1h_64_irq_handler+0x18/0x24
2024-09-26T11:32:36.067249+00:00 ubuntu kernel:  el1h_64_irq+0x74/0x78
2024-09-26T11:32:36.067250+00:00 ubuntu kernel:  sdhci_cqe_enable+0x130/0x150
2024-09-26T11:32:36.067253+00:00 ubuntu kernel:  cqhci_request+0x148/0x460
2024-09-26T11:32:36.067255+00:00 ubuntu kernel:  mmc_cqe_start_req+0x6c/0x104
2024-09-26T11:32:36.067256+00:00 ubuntu kernel:  mmc_blk_mq_issue_rq+0x234/0x3ac
2024-09-26T11:32:36.067258+00:00 ubuntu kernel:  mmc_mq_queue_rq+0x19c/0x21c
2024-09-26T11:32:36.067260+00:00 ubuntu kernel:  blk_mq_dispatch_rq_list+0x234/0x39c
2024-09-26T11:32:36.067262+00:00 ubuntu kernel:  __blk_mq_sched_dispatch_requests+0xc0/0x14c
[... 426 lines cut ...]
2024-09-26T11:32:41.841740+00:00 ubuntu kernel: mmc0: cqhci: Failed to halt
2024-09-26T11:32:41.841770+00:00 ubuntu kernel: mmc0: running CQE recovery
2024-09-26T11:32:42.348472+00:00 ubuntu kernel: mmc0: cqhci: Failed to halt
2024-09-26T11:32:42.365103+00:00 ubuntu kernel: mmc0: running CQE recovery
2024-09-26T11:32:42.881790+00:00 ubuntu kernel: mmc0: cqhci: Failed to halt
2024-09-26T11:32:50.268454+00:00 ubuntu kernel: mmc0: running CQE recovery
2024-09-26T11:32:50.775083+00:00 ubuntu kernel: mmc0: cqhci: Failed to halt
2024-09-26T11:32:50.775119+00:00 ubuntu kernel: mmc0: running CQE recovery
2024-09-26T11:32:51.281771+00:00 ubuntu kernel: mmc0: cqhci: Failed to halt
2024-09-26T11:32:51.281790+00:00 ubuntu kernel: mmc0: running CQE recovery
2024-09-26T11:32:51.788383+00:00 ubuntu kernel: mmc0: cqhci: Failed to halt
2024-09-26T11:32:51.788399+00:00 ubuntu kernel: mmc0: running CQE recovery
2024-09-26T11:32:52.295097+00:00 ubuntu kernel: mmc0: cqhci: Failed to halt
2024-09-26T11:32:52.295129+00:00 ubuntu kernel: I/O error, dev mmcblk0, sector 4047560 op 0x1:(WRITE) flags 0x9800 phys_seg 21 prio class 2
2024-09-26T11:32:52.295132+00:00 ubuntu kernel: Aborting journal on device mmcblk0p2-8.
2024-09-26T11:32:52.298408+00:00 ubuntu kernel: EXT4-fs (mmcblk0p2): failed to convert unwritten extents to written extents -- potential data loss!  (inode 257144, error -30)
2024-09-26T11:32:52.298426+00:00 ubuntu kernel: EXT4-fs error (device mmcblk0p2): ext4_journal_check_start:83: comm rs:main Q:Reg: Detected aborted journal
2024-09-26T11:32:52.298428+00:00 ubuntu kernel: EXT4-fs error (device mmcblk0p2): ext4_journal_check_start:83: comm journal-offline: Detected aborted journal
2024-09-26T11:32:52.301727+00:00 ubuntu kernel: EXT4-fs error (device mmcblk0p2): ext4_journal_check_start:83: comm systemd-journal: Detected aborted journal
2024-09-26T11:32:52.301738+00:00 ubuntu kernel: EXT4-fs error (device mmcblk0p2): ext4_journal_check_start:83: comm .NET TP Worker: Detected aborted journal
2024-09-26T11:32:52.301738+00:00 ubuntu kernel: EXT4-fs (mmcblk0p2): Remounting filesystem read-only
2024-09-26T11:32:52.301740+00:00 ubuntu kernel: EXT4-fs (mmcblk0p2): Remounting filesystem read-only
2024-09-26T11:32:52.301746+00:00 ubuntu kernel: EXT4-fs (mmcblk0p2): ext4_writepages: jbd2_start: 9223372036854775769 pages, ino 2037; err -30
2024-09-26T11:32:52.304982+00:00 ubuntu kernel: audit: type=1400 audit(1727350372.300:52): apparmor="DENIED" operation="open" profile="rsyslogd" name="/run/systemd/sessions/" pid=881 comm=72733A6D61696E20513A526567 requested_mask="r" denied_mask="r" fsuid=102 ouid=0
```

tomharvey, 2024-09-26:
After a reboot I ended up in busy box.

After running fsck -y on the device I can exit out of busy box and all is well again... I assume, until the next time the queue gets borked

HeyMeco, 2024-09-30:
> After a reboot I ended up in busy box.
> 
> After running fsck -y on the device I can exit out of busy box and all is well again... I assume, until the next time the queue gets borked

Had the same thing happen. The above fix also worked on mine

Joshua-Riek, 2024-10-10:
Interesting, dropping the eMMC down to HS200 is the proposed change here? 

While this is not a fix, dropping https://github.com/Joshua-Riek/linux-rockchip/commit/d03a5046f13a8a5b4d7c4b02e442aa4508890413 may be a better option, but how will this affect other devices?

HeyMeco, 2024-10-10:
@Joshua-Riek 
> how will this affect other devices?

Should only reduce IOPS a bit but better to have a stable device

nyanmisaka, 2024-10-11:
As an alternative, add these only to known affected devices?
```c
&sdhci {
...
    /delete-property/ supports-cqe;
...
};
```

HeyMeco, 2024-10-27:
I think I need to try a dtbo like this on Rock5B:
```
/dts-v1/;
/plugin/;

/ {
	fragment@0 {
		target = <&sdhci>;
		__overlay__ {
			mmc@fe2e0000 {
				/* Commenting out supports-cqe */
				// supports-cqe;
			};
		};
	};
};
```

closed 2025-02-10 by github-actions[bot]