Joshua-Riek/ubuntu-rockchip #1072: Bug Report: Orange pi 5 / Microphone at Chrominium not working

dsntmria, 2024-10-03:
### What happened?

The microphone its working fine for all applications mostly but its glitching with Chromium.. or any other browser.. any thoughts? 

### Kernel version

6.1.0-1020-rockchip

### SBC model

Orange Pi 5

### What operating system are you seeing this problem on?

Ubuntu 22.04 LTS (Jammy Jellyfish)

### Relevant logs

```shell
no logs but its glitching the mic icon at the top bar. And the mic is not working.
```

Joshua-Riek, 2024-10-03:
Welcome to the world of Rockchip devices, get an actual computer if you expect this to work.

closed 2024-10-03 by Joshua-Riek

dsntmria, 2024-10-03:
Thank you, at least I know what to expect now. 
reasons why?

Joshua-Riek, 2024-10-03:
Ill be making a detailed statement later this month. But there is no funding, i do not get paid for this crap, none of the 9+ companies ive had detailed conversations with will support me, I do not have the hardware, or equipment to continue development. And when I ask companies for help, ive been insulted multiple times on a far to personal level. So im at the point where I do not care, I'm pretty close to deleting the entire GitHub project. I blame Orange Pi partially as they are the ones who attacked me. Even though I provide way better software for there products for free and have well over 100K downloads alone for Orange Pi.

dsntmria, 2024-10-04:
I wanted to take a moment to express my appreciation for your amazing project. Although I am not in a position to support it financially at the moment, I hope to contribute in the future. In the meantime, I have been using my Orange Pi 5 to explore and experiment with your project. Unfortunately, my everyday computer's power supply burned out, so I have been working from the Orange Pi with Ubuntu, which is why I need to use the browser's microphone. Please keep up the great work and never discontinue this amazing project.