Joshua-Riek/ubuntu-rockchip #1073: Bug Report: Orange Pi 5 Max UART2 M0 Device Tree Overlay

joeyboey, 2024-10-04:
### What happened?

### Razberry Zwave UART Adapter

I tried moving my Home Server Setup to the Orange Pi 5 Max Board. Unfortunately i can't get the UART2 [Port](http://www.orangepi.org/img/pi5-max/max-15.png) to work.

I added the following line to the u-boot file according to the [manual](https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-24.04-LTS#using-a-device-tree-overlay):

```
U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/rk3588-uart2-m0.dtbo"
```

Everything else is commented out, the full file looks like this:
```
## /etc/default/u-boot - configuration file for u-boot-update(8)

#U_BOOT_UPDATE="true"

#U_BOOT_ALTERNATIVES="default recovery"
#U_BOOT_DEFAULT="l0"
#U_BOOT_PROMPT="1"
#U_BOOT_ENTRIES="all"
#U_BOOT_MENU_LABEL="Debian GNU/Linux"
#U_BOOT_PARAMETERS="ro earlycon"
#U_BOOT_ROOT=""
#U_BOOT_TIMEOUT="50"
#U_BOOT_FDT=""
#U_BOOT_FDT_DIR="/lib/firmware/"
#U_BOOT_FDT_OVERLAYS=""
#U_BOOT_FDT_OVERLAYS_DIR="/lib/firmware/"
#U_BOOT_SYNC_DTBS="false"
U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/rk3588-uart2-m0.dtbo"
```

The reboot after applying the chages fails until i remove the Razberry Adapter. The Pi is just stuck on the teal Power LED.
The Pi does boot normally when the UART Adapter is removed.
The Adapter ist working fine, as it was in constant use just before the switch.

Thanks for your help!

### Kernel version

6.1.0-1025-rockchip

### SBC model

Orange Pi 5 Max

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

martin-0, 2025-01-26:
In my opi5 max I used both precompiled image (24.04) and built it myself using Joshua's built script - it worked without problem.
Similarly when I was just messing around and compiled my own uboot (both recent 2025 repo) and with custom dts provided by [Jimmy Hon](https://patchwork.kernel.org/project/linux-arm-kernel/patch/20250109051619.1825-5-honyuenkwun@gmail.com/). In all cases uart2 was working as expected.

It's worth noting that not all programs are happy with 1500000 baud rate (e.g. cu), I used minocom. Also not all adapters can push this speed.

jimmyhon, 2025-02-21:
Try keeping the `chosen` in the DTS if you want to use it as a console.

https://github.com/Joshua-Riek/linux-rockchip/blob/noble/arch/arm64/boot/dts/rockchip/rk3588-orangepi-5-max.dts#L29

The mainline has that node left in.
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/arm64/boot/dts/rockchip/rk3588-orangepi-5.dtsi?h=v6.14-rc1#n22
...because the Orange Pi 5 Plus originally had it in there.