Joshua-Riek/ubuntu-rockchip #1084: Bug Report: OPI5B V2.3.2 No HW ACC for Moonlight

duruer, 2024-10-10:
### What happened?

Hi, I installed on orangepi 5B, first of all it was broken Bluetooth. Then after updating system with pop-up it opened first, it got fixed after reboot.

Anyway, I installed moonlight-qt but says no HW acceleration. I thought this preinstalled image supposed to have support for HW acceleration by default? or am I doing something wrong?

Thank you in advance.

### Kernel version

6.1.0-1020-rockchip

### SBC model

Orange Pi 5B

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

konstantinGlonin, 2025-01-22:
Hi @duruer 
Cab you fix the problem?

duruer, 2025-01-24:
Hi, unfortunately can not.

closed 2025-08-25 by github-actions[bot]