Joshua-Riek/ubuntu-rockchip #1088: Bug Report: Rock 5b+ Ubuntu FAN is not spining

tpnthr, 2024-10-13:
### What happened?

So the problem is that fan is not spinning. It is original fan  with heat sink from radxa.
SBC is getting pretty hot, but fan is still. When i turn on the device fan moves a bit and is still for entire time.

sudo sh -c 'echo 255 > /sys/class/hwmon/hwmon9/pwm1'; echo $?
shows 0 error and parameter is changed to 255 for a few seconds, but after that parameter is set back to 0. fan do not react to any type of changes.

Maybe someone knows how to make this fan work?

### Kernel version

6.1.0-1025-rockchip

### SBC model

radxa 5b+

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

lukaszsobala, 2024-10-16:
To make the manual fan speed be persistent, you need to change the policy to userspace:

`echo user_space |sudo tee /sys/class/thermal/thermal_zone0/policy`

The fan does not spin because the default profile does not make sense. I recommend this one:
https://forum.radxa.com/t/guide-how-to-make-the-rock-5b-fan-work-properly-joshua-rieks-ubuntu-24-04/20836

tpnthr, 2024-10-16:
Done everything according to that instructions
Still nothing.
When I plug power to sbc FAN make one rotation and stops - so it should be connected good. 

May be there is a way to force fan to certain speed?

lukaszsobala, 2024-10-16:
the way to force a certain speed that I know about is echoing some number to pwm1, as you did, with user_space policy. Or you can try powering it with GPIO if you know how to use them.

cplatt-iso, 2024-10-28:
On my stock installation the pwm fan is hwmon10 every boot.  You can check the value of  /sys/class/hwmon/hwmonX/name where X is some number.  One of them will be "pwmfan".

You need to make sure the correct device tree is loaded, see: [https://github.com/Joshua-Riek/ubuntu-rockchip/issues/1090](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/1090)

You dont need to set user_space policy out of the box, "echo 255 | sudo tee /sys/class/hwmon/hwmon10/pwm1" works for me and set the fan at 100% duty cycle.    There is an issue with the device tree and my fan, where it is setting a duty cycle of 64 which does not spin my fan up.  I've made a few changes in my file if you want to try it: [https://github.com/cplatt-iso/rock5b/blob/master/rk3588-rock-5b-plus.dts](https://github.com/cplatt-iso/rock5b/blob/master/rk3588-rock-5b-plus.dts)

closed 2025-02-10 by github-actions[bot]