Joshua-Riek/ubuntu-rockchip #1090: Bug Report: Rock 5B Plus does not boot with the correct device tree

cplatt-iso, 2024-10-15:
### What happened?

When booting a fresh installation of this distro, the device-tree model is incorrectly selected as "Radxa ROCK 5B"
`root@ubuntu:~/dtb# strings /proc/device-tree/model
Radxa ROCK 5B
`
Using release: v2.3.2/ubuntu-24.04-preinstalled-server-arm64-rock-5b-plus.img.xz

This results in the following lspci with a drives present in both nvme slots:
`ubuntu@ubuntu:~$ lspci
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
0000:01:00.0 Non-Volatile memory controller: Micron/Crucial Technology T500 NVMe PCIe SSD (rev 01)
0002:20:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
0002:21:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8852BE PCIe 802.11ax Wireless Network Controller
0004:40:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
0004:41:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
`

When loading the proper device tree file this issue is resolved;
`ubuntu@ubuntu:/etc/default$ cat /proc/device-tree/model
Radxa ROCK 5B Plus`

`ubuntu@ubuntu:/etc/default$ lspci
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
0000:01:00.0 Non-Volatile memory controller: Micron/Crucial Technology T500 NVMe PCIe SSD (rev 01)
0001:10:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
0001:11:00.0 Non-Volatile memory controller: Micron/Crucial Technology T500 NVMe PCIe SSD (rev 01)
0002:20:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
0002:21:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8852BE PCIe 802.11ax Wireless Network Controller
0004:40:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
0004:41:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
`

I resolved this by editing the extlinux.conf file and setting:
`fdt /lib/firmware/6.1.0-1025-rockchip/device-tree/rockchip/rk3588-rock-5b-plus.dtb` 

I presume this is not the way to do this properly and u-boot should be fixed or something along those lines.  However, everything does work when the proper device-tree file is loaded.

### Kernel version

6.1.0-1025-rockchip

### SBC model

Radxa ROCK 5B Plus

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

ewaldc, 2025-01-08:
It is probably a defect, but since there is no more support for this distribution it's very likely not going to be fixed.  That said, you have issued the right fix to load the proper Device Tree Overlay.  U-boot just loads `/boot/extlinux/extlinux.conf` so there is nothing to be done at `u-boot` level itself.  Well done!

MarcA711, 2025-01-18:
It seems that this issue was fixed. At least the `cat /proc/device-tree/model` shows `Radxa ROCK 5B Plus` for me after installing the 2.4.0 release. Maybe this issue can be closed to avoid confusing new users.

closed 2025-03-24 by github-actions[bot]