Joshua-Riek/ubuntu-rockchip #1093: Feature Request: Add rknpu version 0.9.8

vincenzodentamaro, 2024-10-17:
### Feature Description

As per the title, please add the new RKNPU kernel version 0.9.8 contained here https://github.com/airockchip/rknn-llm/tree/main/rknpu-driver

In the documentation, rockchip says that if the build fails modify the kernel header file kernel/include/linux/mm.h by adding the following code: 
```

static inline void vm_flags_set(struct vm_area_struct *vma, 
vm_flags_t flags){ 
vma->vm_flags |= flags; 
} 
static inline void vm_flags_clear(struct vm_area_struct *vma, 
vm_flags_t flags){ 
vma->vm_flags &= ~flags; 
} 
```

please update the releases with such kernel driver ASAP. If possible compile it as external module to be loaded with modprobe so if they update the driver, we can recompile it and load it with modprobe instead of recompiling the kernel.

Thank you

Joshua-Riek, 2024-10-17:
I do not have time, can you open a pull request?

vincenzodentamaro, 2024-10-17:
Yes of course, I will modify the kernel and do a pull request.
Thank you for answering

darkice9x, 2024-10-23:
vm_flags_set and vm_flags_clear have already been added to ./linux-rockchip/include/linux/version_compat_defs.h, and there are functions that were not implemented when adding rknpu 0.9.8, but they are not needed when using rk3588, so just comment them out.

To comment out the specific line in linux-rockchip/drivers/rknpu/rknpu_devfreq.c at line #237 .set_soc_info = rockchip_opp_set_low_length,.

c0zaut, 2024-11-02:
@SimplyCorbett - it was submitted via the kernel repo: https://github.com/Joshua-Riek/linux-rockchip/pull/36

closed 2025-02-10 by github-actions[bot]