Joshua-Riek/ubuntu-rockchip #1099: Feature Request: Remove lxd snap from the image

rmbleeker, 2024-10-21:
### Feature Description

To my surprise the snap `lxd` was already installed after flashing the image. Removing this snap package and it's `core22` dependency will make the download smaller and the flashing process faster, not to mention it will please those people who dislike how Canonical is forcing their snap ecosystem on their users. 

I personally don't have an issue with snaps, but I do think a base image should be as lean as possible. `lxd` is very specific software that not everyone installing this might want to use, so having it installed by default is just bloat and possibly another attack vector.

Joshua-Riek, 2024-10-21:
I'm using the official Ubuntu build system and do not want to deviate greatly from traditional Ubuntu images. The only exceptions where I remove pre-installed snaps are for desktop images, where I'm constricted by a 2GB artifact limit on GitHub.

rmbleeker, 2024-10-21:
That's fair, I didn't know Canonical started shipping `lxd` with their builds by default. In that case I'll close this issue.

closed 2024-10-21 by rmbleeker