Joshua-Riek/ubuntu-rockchip #1100: Feature Request: How can I use GPIO or UART on these images?

czf0613, 2024-10-21:
### Feature Description

My board is OrangePi 5 pro with RK3588S.
I cloned the official wiringOP and built the lib. Then I run the `sudo gpio readall` command, it shows that wiringOP cannot identify the board version due to lack of some files like '/etc/orange-release'.

By the way, how can I enable the UART device on my board? On other systems like Armbian, I can add overlays to `/boot/armbianEnv.txt` to enable the UART device, but I could not find it on these ubuntu images.

I am sincerely waiting for solution.

Joshua-Riek, 2024-10-21:
https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-24.04-LTS#using-a-device-tree-overlay

czf0613, 2024-10-21:
I see the `U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/rk3588-i2c0-m1.dtbo"`.
If I have more than one overlays, what should I do?

Also, I could not find any `.dtbo` file in the file system, should I prepare it myself?

darkice9x, 2024-10-22:
# Add the following text to /etc/default/u-boot on Orange Pi 5
U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/rk3588-uart0-m2.dtbo device-tree/rockchip/overlay/rk3588-uart1-m1.dtbo"
# Add the following text to /etc/default/u-boot on Orange Pi 5 Plus
U_BOOT_FDT_OVERLAYS_DIR="/lib/firmware/6.1.0-1016-rockchip/device-tree/rockchip/overlay/"
U_BOOT_FDT_OVERLAYS="rk3588-uart3-m1.dtbo rk3588-uart4-m2.dtbo"

closed 2024-10-22 by czf0613