Joshua-Riek/ubuntu-rockchip #1102: Bug Report: rock5b network wlan devices missing/wrong driver

schnellert, 2024-10-22:
### What happened?

I have the network manager installed with:
`sudo apt-get update;`
`sudo apt-get install network-manager -y`

and then the devices are listed there:
`nmcli d`
[image]

-
My problem is that there is no wlan network here. With a normal Raxda Ubuntu image i have 'wlan0' and 'wlan1', here i only have 'wlP2p33s0'

### Kernel version

6.1.0-1023-rockchip

### SBC model

Radxa ROCK 5B

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

Joshua-Riek, 2024-10-22:
Ubuntu 24.04 uses netplan, do not use network manager.

schnellert, 2024-10-23:
In netplan i actually only get one displayed.

network-manager is not intalled. is a fresh ubuntu 24.04

[image]

schnellert, 2024-10-23:
Do i need a different configuration for netplan?

schnellert, 2024-10-29:
Any ideas?

closed 2025-02-10 by github-actions[bot]