Joshua-Riek/ubuntu-rockchip #1113: Bug Report: No sound orange pi5 ubuntu 24.10

ltore1234, 2024-10-29:
### What happened?

sound output device dummy output  only no hdmi nor analog  installed pülse audio no luck   No sound orange pi5 ubuntu 24.10 kernel 6.11

### Kernel version

6.11.0

### SBC model

Orangepi 5

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

### Relevant logs

```shell
no logs
```

ltore1234, 2024-10-29:
sorry  i mean ubuntu 24.10

Joker41NAM, 2024-11-03:
Similar issue; Orange Pi 5 Plus, Ubuntu 24.10.  No HDMI audio detected, just Analog Out (I don't have a set of speakers handy to test whether the Analog is actually working)

nrpetonr, 2024-11-04:
Full HDMI support has not been merged into the 6.11 Linux kernel, so expect it not to work. If it does work, it will have limited support and functionality.