Joshua-Riek/ubuntu-rockchip #1119: Bug Report: Turing RK1 and USB

CharlesPerrotMinotHCHB, 2024-11-04:
### What happened?

Hi,

I'm using a Turing Pi 2, on which I have a Turing RK1.
Tried both the 24.04 and 24.10 Ubuntu you guys have released, and none work with the USB (I have a mouse plugged in, but the cursor doesn't move, which makes any setup impossible). On the other hand, the official image for 22.04 (https://firmware.turingpi.com/turing-rk1/ubuntu_22.04_rockchip_linux/v1.33/) support USB without issues.

Thanks!

### Kernel version

6.1 and 6.11

### SBC model

Turing RK1

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

SimonStiil, 2024-11-04:
Just flashed the latest [v2.4.0](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/tag/v2.4.0) Ubuntu 24.04 Server 
ubuntu-24.04-preinstalled-server-arm64-turing-rk1.img.xz 
Image to my turing pi and I have no issues with it, Both HID Devices on First board where keyboard worked without a hitch and Super Speed Mass storage controller on the 4th board supporting that.

CharlesPerrotMinotHCHB, 2024-11-05:
Appreciate @SimonStiil , but that's what I did and it didn't work at all.
- I flashed from the UI, how did you?
- Tried both the OTG usb as well as the other two, both with the card in `host` or `device` mode, same results

SimonStiil, 2024-11-05:
@CharlesPerrotMinotHCHB I flashed it from the Turing PI 2 BMC Web UI as well on all 4 boards,
Downloaded the xz zipped image. Unziped the image to the img file. 
Selected the image in the Web UI and upladed it via the web interface.

Daemon version:
2.0.5
Build version:
2023.08

USB configuration for White USB2 port is Device mode connected to Node 1. 

Only once had a use for any of the other modes but failed to get them to work correctly for me as i did not have any other Native Linux PC's in my lab.

CharlesPerrotMinotHCHB, 2024-11-06:
That's exactly what I did, and it never worked for me, only with the official images from Turing org. Will retry on Monday just in case, but it's also pretty hard to debug, since it just doesn't work

closed 2025-02-10 by github-actions[bot]

joekplee, 2025-02-27:
Recently,  I had the same issue to flash CM4 on nodes.  Have already set flash on specific node and had already ran the rpiboot -d .\msd command  and then plug-in usb-A cable.  No responses at all.  It seems to me that the USB 2.0 have no responses at all.  Any ideas how it is possible to check the status of the usb2.0 for the assurance of the connectivity? Please advise and thank you.

CharlesPerrotMinotHCHB, 2025-05-10:
Unfortunately, they autoclosed the issue, even though no help was really provided, @joekplee so don't expect anything here