Joshua-Riek/ubuntu-rockchip #1125: Bug Report: Orange Pi 5 Pro looses bluetooth pairing after warm reboot 22.04)

bartekuk, 2024-11-23:
### What happened?

I am able to pair my mouse/keyboard with the Orange Pi 5 Pro running 22 Ubuntu just fine but on warm reboot the pairing is lost confirmed with bluetoothctl. I will look into it further and update ticket tomorrow when I get usb BT to experiment with. I will also provide more logs and /var/lib/bluetooh snapshots. That is just to start the ball rolling

And besides great thanks for this effort I find these images far better then Linux images provided by Orange Pi

### Kernel version

5.10-1022-rockchip

### SBC model

Orange Pi 5 Pro

### What operating system are you seeing this problem on?

Ubuntu 22.04 LTS (Jammy Jellyfish)

dgdonovan, 2024-11-24:
See https://github.com/Joshua-Riek/ubuntu-rockchip/discussions/1104

bartekuk, 2024-11-24:
> See #1104

Oh bummer. Shame on Rockchip for not supporting such promising endeavour

jinglong485, 2024-11-30:
I had the same issue in 24.04, deleting --bd_addr_rand in usr/lib/systemd/system/ap6256-bluetooth.service solved my problem.

closed 2025-02-10 by github-actions[bot]

default-writer, 2025-10-17:
Orange Pi 5 Max solution:

I am using service to run this command to up and run Bluetooth on login screen:

```
dbus-send --system --type=method_call --print-reply=literal --dest=org.bluez /org/bluez/hci1 org.freedesktop.DBus.Properties.Set string:org.bluez.Adapter1 string:Powered variant:boolean:true
```

also had run this command:
```
echo 'mmc2:0001:3' | sudo tee /sys/bus/sdio/drivers/btsdio/unbind
```

and to make this `bt-startup.service`  to run:

```
sudo nano /etc/systemd/system/bt-startup.service

```
and put this:

```
[Unit]
Description=Bluetooth power keeper
After=network.target

[Service]
Type=oneshot
RemainAfterExit=yes
ExecStart=/usr/bin/dbus-send --system --type=method_call --dest=org.bluez /org/bluez/hci1 org.freedesktop.DBus.Properties.Set string:org.bluez.Adapter1 string:Powered variant:boolean:true
ExecStop=/usr/bin/dbus-send --system --type=method_call --dest=org.bluez /org/bluez/hci1 org.freedesktop.DBus.Properties.Set string:org.bluez.Adapter1 string:Powered variant:boolean:false

[Install]
WantedBy=multi-user.target
```

and run it:
```
sudo systemctl enable bt-startup.service
sudo systemctl start bt-startup.service

```