Joshua-Riek/ubuntu-rockchip #1133: Bug Report: RGA 4G limit memory

MattiaGrigoli, 2024-11-28:
### What happened?

on higher memory board (+4g of RAM) the RGA stops working becuase of the limit of RGA2 of 4G caused by its 32 bits addresses.
is there a fix for this issue? i need to use the RGA for professional reasons and i can't find a solution anywhere

### Kernel version

5.10/6.10

### SBC model

orange pi 5 plus

### What operating system are you seeing this problem on?

Ubuntu 22.04 LTS (Jammy Jellyfish)

nyanmisaka, 2024-11-29:
Check https://github.com/Joshua-Riek/ubuntu-rockchip/issues/1030#issuecomment-2365178523.
The RGA2 hardware is designed in such a way that you must use the DMA32 flag when allocating memory.

So this is not an issue and @Joshua-Riek should feel free to close these two.

unitedtimur, 2025-01-10:
> Check [#1030 (comment)](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/1030#issuecomment-2365178523). The RGA2 hardware is designed in such a way that you must use the DMA32 flag when allocating memory.
> 
> So this is not an issue and @Joshua-Riek should feel free to close these two.

After using  "/dev/dma_heap/system-uncached-dma32", when the call becomes very frequent, for example about 60 allocations per second, and about the same for deallocation, then at some point I get the error RK_DMA_HEAP_ALLOC_BUFFER failed in dma_buf_alloc

And I do not know how to fix it. Can you help with this?

nyanmisaka, 2025-01-10:
> After using "/dev/dma_heap/system-uncached-dma32", when the call becomes very frequent, for example about 60 allocations per second, and about the same for deallocation, then at some point I get the [...]

This is a rare use case and I haven't had a chance to encounter it. Clients should **reuse** allocated DMA buffers as much as possible, instead of always **freeing + reallocing** them.

Also, rockchip has retired their custom dma-heap allocator in 6.1 LTS kernel.

If you insist that this is an issue, please write to rockchip for more info.
https://opensource.rock-chips.com/wiki_Main_Page

closed 2025-03-17 by github-actions[bot]