Joshua-Riek/ubuntu-rockchip #1148: Feature Request: Please add support for Orange Pi 5 Ultra

BlindWanderer, 2024-12-17:
### Feature Description

The only difference between the Orange Pi 5 Ultra and the Max is that the Max has two HDMI Out and the Ultra lacks HDMI_0 and instead has the standard HDMI_RX.

I will happily test any build. Thank you for your hard work on this project.

therysin, 2024-12-17:
Also requesting. Thanks for all the work you've put in on this project.

yan00s, 2024-12-23:
Also requesting. Thanks a lot

yan00s, 2025-02-15:
not outdated!

VoidLock, 2025-04-06:
Also Requesting

nstilt1, 2025-04-21:
Also requesting. Trying to use this board for my masters project (it is due next year, so plenty of time). I have a working OPi 5 prototype but it has a lot of latency due to it needing an HDMI capture card. I will try to look through the PRs and commits that added support for previous boards to try to figure out what needs to be added, but it could take me a few months to piece everything together. I would greatly appreciate some insight into the support process, but if nothing else, I'll just try to use all the resources I have access to (github, LLMs). And before I forget, thank you for making this repo. Your OPi 5 image worked wonders while the official Ubuntu image did not.

DrHo1y, 2025-04-24:
Also Requesting

samtux, 2025-08-06:
also requesting

MarvinVillagra, 2025-08-08:
move to radxa, it is better and it is working

labmaster, 2025-09-03:
also requesting

mrspo, 2025-11-08:
also requesting. there aren't a lot of options for this board.

i'm shopping around for a new OS because i have a feeling the orange pi 5 max armbian image i've been using (which others have reported as 'fine') is causing the crazy cpu thrashing issues i've been seeing. Not to mention the NVMe disk is not getting the same performance as my other 5 ultra. it's the boot disk so maybe there's some jiggery pokery i need to do, i haven't looked into it yet.

the other one runs OPI OS minimal bookworm but i stripped out the default OPI apt sources and replaced with the standard Debian sources aaaaand i don't think this thing has actually gotten any updates in a long time, so i'll be reinstalling something on that one as well.

i'm just a guy trying to run some tiny servers. the only thing i see being maintained for opi 5 ultra is dietpi, which is what i'll try next. i'd love to try this ubuntu build too, just not until there's a dedicated image