Joshua-Riek/ubuntu-rockchip #1151: Feature Request: Where are the instructions on how to build this please ?

AaronNGray, 2024-12-22:
### Feature Description

I have tried to build this with 
```
sudo ./build.sh --board=armsom-sige7
```
And get the following :--project=preinstalled-desktop --release=jammy --kernel=rockchip-5.10
```
And get nothing happening.

nobara256, 2024-12-24:
You need to build it like `sudo ./build.sh --board=orangepi-5-plus --suite=oracular --flavor=server`

where

```
--suite=[oracular|jammy|noble]
--flavor=[server|desktop]
```

How ever I would recommend you to build the kernel and u-boot package separately first with `--kernel-only` and `--uboot-only` as they take additional time.

closed 2025-02-24 by github-actions[bot]

gjrtimmer, 2025-03-19:
what are the build dependencies when running this on a newly installed distro?