Joshua-Riek/ubuntu-rockchip #1157: Attempting to get 10 inch LCD touch to work

atlaste, 2024-12-28:
### What happened?

Trying to get the official 10" LCD from Orange PI to work

### Kernel version

5.10.0-1012-rockchip

### SBC model

Orange PI 5 MAX

### What operating system are you seeing this problem on?

Ubuntu 22.04 LTS (Jammy Jellyfish)

### Relevant logs

```shell
First off, I'd like to thank Riek for his work. I'm also leaving this for reference, so other people can benefit from this. My goal here is to have the OPI 5 running in Kiosk mode. 

# Current status:

Display, X11, kiosk mode work.
Goodix touch panel does not work.

## What I did:

I've gathered information from various places here, so I though it would be helpful to make some remarks on how to do this.

For the wiring, check out the manual from the OPI website. **Don't look at the silkscreen**, because apparently you need to put pin 40 in the connector at the spot of pin 1. Well whatever...

Next, DO NOT grab the latest kernel. It just doesn't work with the LCD panels. You need the older kernel version. You can spend days troubleshooting this, like I did.

Next, you need to add some overlays to get this working. Edit /etc/defaults/u-boot with your favorite editor and add:

U_BOOT_FDT="/usr/lib/firmware/5.10.0-1012-rockchip/device-tree/rockchip/rk3588-orangepi-5-max.dtb"
U_BOOT_FDT_OVERLAYS="orangepi-5-max-disable-leds.dtbo orangepi-5-max-lcd.dtbo rk3588-i2c7-m3.dtbo"
U_BOOT_FDT_OVERLAYS_DIR="/usr/lib/firmware/5.10.0-1012-rockchip/device-tree/rockchip/overlay"

Obviously, not everyone needs the '-max' but something else- go to the overlay folder and check it out.

The disable-leds is obviously for disabling the annoying LED. The I2C7 was my attempt to enable the I2C port for the touch panel; apparently this is a Goodix GT9271 IC, which runs through I2C. From the MAX datasheet (OPI website) I found the schematic from the board, and from that you can gather the I2C port.  Pin 23/24 sais I2C7_SCL_M0 / I2C7_SCA_M0 -- which implies I2C port 7.

Reboot and see your LCD panel come to live. Great! But we're obviously not there yet.

That being done, you can go ahead and install X and setup a kiosk. Like: https://blog.r0b.io/post/minimal-rpi-kiosk/ . Other than that, put a line in `/etc/X11/Xwrapper.config`: `allowed_users=anybody` and run startx. 

At this point, you have a kiosk running with Chromium.

## Touch screen

The touch screen is apparently a Goodix GT9271 touch screen. So from what I gathered, there's a linux driver named "gt9xx" which contains the relevant code. In `/usr/lib/modules/5.10.0-1012-rockchip` I found that `modules.builtin` contains `kernel/drivers/input/touchscreen/gt9xx/goodix_gt9xx.ko`... but unfortunately its not picked up by the kernel for some reason.

If anyone knows the answer, please let me know.
```

atlaste, 2024-12-28:
Just some minor progress...

1. Apparently the `rk3588-i2c7-m3.dtbo` overlay is not needed. It doesn't seem to do much harm tho.

2. `/sys/firmware/devicetree/base/i2c@fec90000/touchscreen@14` has a `compatible` file with the string "goodix,gt9271". Just to be sure, let's see if it's also found as an i2c device? The name is apparently "touchscreen".

`i2cdetect -a 7` -> 0x14 is found, which is correct.

disruptivepatternmaterial, 2025-01-16:
Just trying to learn here...so I am using 6.11.0-1006-rockchip and there is no ......device-tree/rockchip/overlay/rock-5b-radxa-display-8hd.dtbo guessing from your bug here, and the lack of this file in the 6.11 here, I am guessing that I need to go back a release to find these files to make the display work ... even to get as far as you have? Or is there a way to generate these files for the newer kernel?

closed 2025-03-17 by github-actions[bot]