Joshua-Riek/ubuntu-rockchip #1161: Bug Report: Orange Pi 5 Pro can't flash u-boot on SPI

aik3dev, 2025-01-03:
### What happened?

Hi, when I try to laminate u-boo” on SPI I get the error “u-boot-install-mtd: block device ‘/dev/mtd0’ not found”. My system does not detect this device and thus I cannot boot from the nvme drive. Please help me
[image]

### Kernel version

6.1.0-1025-rockchip

### SBC model

Orange Pi 5 Pro

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

Plastic-armyman, 2025-01-04:
> ### What happened?
> Hi, when I try to laminate u-boo” on SPI I get the error “u-boot-install-mtd: block device ‘/dev/mtd0’ not found”. My system does not detect this device and thus I cannot boot from the nvme drive. Please help me ![Zrzut ekranu z 2025-01-03 17-18-17](https://private-user-images.githubusercontent.com/151382648/400000947-fb2d1931-f9be-4e54-8e28-efa11688af23.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYwMjg4MTAsIm5iZiI6MTczNjAyODUxMCwicGF0aCI6Ii8xNTEzODI2NDgvNDAwMDAwOTQ3LWZiMmQxOTMxLWY5YmUtNGU1NC04ZTI4LWVmYTExNjg4YWYyMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwNFQyMjA4MzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jOWMwN2QwMzAxYmZmZWY3ZjUzZDg3ZWM4YzgzYmU0NWFkOTgzNDMwODQ3ZjE0MWQ4MzI5MWRlNmM2NTU4N2ZmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.dZJcz3k0iwSRzTkhHZLrHvMj3JbRVwGncwVcSQvkUsI)
> 
> ### Kernel version
> 6.1.0-1025-rockchip
> 
> ### SBC model
> Orange Pi 5 Pro
> 
> ### What operating system are you seeing this problem on?
> Ubuntu 24.04 LTS (Noble Nombat)
> 
> ### Relevant logs
> _No response_

Orange Pi 5 Pro doesn't have SPI unless installed by you.

Use an SD card instead and follow those instructions.

Plastic-armyman, 2025-01-04:
> Same problem here: <img alt="Снимок экрана 2025-01-04 в 18 03 54" width="396" src="https://private-user-images.githubusercontent.com/26006994/400146851-36c25bba-9af5-497f-8027-54bac05f146f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYwMjg4MTAsIm5iZiI6MTczNjAyODUxMCwicGF0aCI6Ii8yNjAwNjk5NC80MDAxNDY4NTEtMzZjMjViYmEtOWFmNS00OTdmLTgwMjctNTRiYWMwNWYxNDZmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA0VDIyMDgzMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc0NGI0OTMxYjMwODIyYjYzODhkYTAzNzhjY2MzZmYyMDg3NjlkN2M0Yjk5NjM1MDA4MGEzY2MxYjFhMjlmNjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.sDOSKcxQ6WzeinzMx21elnNmnKOBxt3cQtp4zro9hLw">

Same suggestion for you.

Plastic-armyman, 2025-01-05:
> > Orange Pi 5 Pro doesn't have SPI unless installed by you.
> > Use an SD card instead and follow those instructions.
> 
> Sorry, what instructions do you mean?

Read the wiki. There's instructions for flashing uboot to SD card.

You then move your installation over to the M2 SSD

After that you'll need to remove the root partition on the SD card and it'll boot into the M2.

Joshua-Riek, 2025-01-07:
Lol, this device does not come with an SPI, you need to solder it yourself because Orange Pi is cheap.

closed 2025-01-07 by Joshua-Riek