Joshua-Riek/ubuntu-rockchip #1169: Orange Pi 5 Ultra

pillo-pillo, 2025-01-12:
### Feature Description

Hi i did just got the brand new Orange Pi 5 Ultra.
There isn't a Joshua Riek Ubuntu 24.04 image jet fot it.
But I did manage to install the Orange pi 5 Max on my brant new Orange pi 5 Ultra. It's going well. 
Thanks.

pillo-pillo, 2025-01-14:
P.S.
It's freez sometimes on Chromium with all images I did try Official images Orange pi 5 Ultra Debian Bookworm, Ubuntu Jelly 22.04. And yours Ubuntu 24.04 Oragepi 5 Max.

Joshua-Riek, 2025-01-17:
Why do people not understand how to create a pull request.

pillo-pillo, 2025-01-18:
Thanks a lot...
Maybe people not so clever or smart as you too...

All I try to say... is.....
OP5MAX image does work on OP5ULTRA.

mprimeaux, 2025-01-31:
Also, I should have mentioned this in my initial reply, I use the server edition of the images and not the desktop edition since I'm focused on GPU workloads in k8s.

pillo-pillo, 2025-01-31:
Thanks.

Correct I did boot accidentally Orange Pi 5 Pro from SD Card on to my Orange Pi Ultra and it did go fine as well 

Today cloned running Ubuntu from eMMC to NVMe with this native command `ubuntu-rockchip-install `  it's great you can clone any running device to any device. [Read More ](https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-24.04-LTS#install-ubuntu-onto-an-nvme-from-linux)

ErezAlster, 2025-02-09:
> Why do people not understand how to create a pull request.

Do you have a manual for adding a new platform? I would be happy to do it :)

KingFalse, 2025-02-12:
I also need updates on the Orange Pi 5 Ultra.

jim-wyatt, 2025-03-30:
> Why do people not understand how to create a pull request.

If this is a convenience project, then why not just setup proper git submodules so we don't have to repeat other's work? This would also prevent stale patches...

https://github.com/armbian/build/pull/7902

jim-wyatt, 2025-03-30:
> > Why do people not understand how to create a pull request.
> 
> If this is a convenience project, then why not just setup proper git submodules so we don't have to repeat other's work? This would also prevent stale patches...
> 
> [armbian/build#7902](https://github.com/armbian/build/pull/7902)

Turns out it was more convenient to just go to the original source:

```
git clone --depth=1 --branch=main https://github.com/armbian/build
cd build
./compile.sh BOARD=orangepi5-ultra BRANCH=current RELEASE=noble BUILD_MINIMAL=yes BUILD_DESKTOP=yes KERNEL_CONFIGURE=no
```

Joshua-Riek, 2025-03-30:
> > Why do people not understand how to create a pull request.
> 
> If this is a convenience project, then why not just setup proper git submodules so we don't have to repeat other's work? This would also prevent stale patches...
> 
> [armbian/build#7902](https://github.com/armbian/build/pull/7902)

Well, you are rude. I've put thousands of hours into this project and if you do not like something please submit a pull request with the code and changes requested. I'm sorry that I have a full-time job and that this is only a side project where I receive no support from companies to develop or maintain their devices. 

I'm locking and closing this issue, please be respectful.

closed 2025-03-30 by Joshua-Riek