Joshua-Riek/ubuntu-rockchip #1176: Bug Report: OrangePi 5 Plus USB 3.0 ports not working

54601, 2025-01-26:
### What happened?

Both USB 3.0 ports are not working, 2.0 and USB-C ports are fine. I tried to use a wired keyboard and it seems like those ports don't have power. Also, nothing happens in dmesg when I plug or unplug devices from USB 3.0 ports.

I red that this was fixed in July, 2024. However, my system is the latest and I even tried to run apt-get full-upgrade. However, that didn't solve the problem.

lsusb showed these:
```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 003: ID 03f0:036b HP, Inc HP USB-C Dock G5
Bus 007 Device 004: ID 03f0:056b HP, Inc USB Audio
Bus 007 Device 005: ID 03f0:086b HP, Inc USB2734
Bus 007 Device 006: ID 03f0:046b HP, Inc HP USB-C Dock G5
Bus 007 Device 007: ID 046d:c548 Logitech, Inc. Logi Bolt Receiver
Bus 007 Device 008: ID 046d:c547 Logitech, Inc. USB Receiver
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 002: ID 03f0:066b HP, Inc HP USB-C Dock G5
Bus 008 Device 003: ID 03f0:076b HP, Inc USB5734
Bus 008 Device 004: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
```

### Kernel version

6.1.0-1025-rockchip

### SBC model

Orange Pi 5 Plus

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

### Relevant logs

```shell

```

yulian5, 2025-04-28:
The same configuration: actually keyboard/mouse work on front USB 3.0 but nothing else. Google AI provide quite useful hints about the problem:

The USB 3.0 ports on an Orange Pi 5 Plus might not work on Ubuntu 24.04 due to a few potential issues: a bug with the U-Boot vendor, incorrect power supply, or the system not being configured to properly utilize USB 3.0. Updating the kernel, verifying the power supply, and ensuring proper U-Boot configurations are crucial for resolving this issue. 
Here's a more detailed breakdown:
1. Kernel and U-Boot Issues:
Kernel Updates:
Updating the kernel to versions like 6.1.0-1018 or 6.1.0-1019 might resolve some USB 3.0-related bugs, as reported in [a thread on Reddit](https://www.reddit.com/r/OrangePI/comments/1drtrqu/joshuariek_ubunturockchip_2404_crashing_for/). 
U-Boot Configuration:
Some users have reported that the U-Boot provided by Orange Pi for the Orange Pi 5 Plus may not fully support USB 3.0. This could involve needing to manually configure or update the U-Boot environment. 
2. Power Supply Issues:
Insufficient Power: Ensure your power supply is capable of providing enough current to the USB 3.0 ports, especially if you're connecting power-hungry devices like external hard drives. A faulty or weak power supply can cause voltage drops and prevent the USB 3.0 ports from functioning correctly. 
3. System Configuration:
USB Device Compatibility:
Verify that the USB devices you are trying to use are compatible with Linux and the USB 3.0 standard.
Driver Issues:
While unlikely, ensure that the necessary USB 3.0 drivers are installed and configured correctly in Ubuntu 24.04. 
4. Troubleshooting Steps:
sudo journalctl -b -1 -e:
This command can help identify any error messages related to USB devices or the system. 
lsusb command:
This command lists all USB devices connected to your system, which can help verify if the system recognizes the USB 3.0 ports. 
Power Cycle:
Try manually power-cycling the Orange Pi 5 Plus after a fresh flash of the bootloader to the SPINOR, [according to a post on Orange Pi's forum](http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=147271&page=1). 
Check for Manual USB 3.0 Support:
Some users have reported that manually enabling USB 3.0 support in the U-Boot configuration is necessary. 
5. Additional Tips:
Try a Different Power Supply:
If you suspect the power supply, try using a different one to see if it resolves the issue. 
Consult Documentation:
Check the official Orange Pi documentation and community forums for more specific instructions and troubleshooting tips related to USB 3.0 support on the Orange Pi 5 Plus.

yulian5, 2025-05-05:
Maybe it'll help someone:

Ubuntu 24.04 image -  if you have inserted flash drive or SSD in USB 3.0 ports before OS booting then it works until they unmounted. Ubuntu 24.04 cannot detect USB 3.0 devices during runtime (except mouse/keyboard);

Ubuntu 24.10 image - seems to work fine most of the time with USB 3.0 ports (however in may case only one HDMI-1 display works somehow).