Joshua-Riek/ubuntu-rockchip #1183: Bug Report: Orange PI CM5 USB3.0 port not working

spele1100, 2025-02-06:
### What happened?

Similar issue: https://github.com/Joshua-Riek/ubuntu-rockchip/issues/1176
But the 3.0 root hub is not even show in lsusb.

image: ubuntu-24.04-preinstalled-server-arm64-orangepi-cm5.img

### Kernel version

6.1.0-1025-rockchip

### SBC model

Orange PI CM5

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

### Relevant logs

```shell
$ dmesg | grep 3.0
[    5.002783] Linux version 6.1.0-1025-rockchip (buildd@bos03-arm64-034) (aarch64-linux-gnu-gcc (Ubuntu 13.2.0-23ubuntu4) 13.2.0, GNU ld (GNU Binutils for Ubuntu) 2.42) #25-Ubuntu SMP Mon Aug 26 23:01:14 UTC 2024
[    5.423180] percpu: Embedded 29 pages/cpu s80504 r8192 d30088 u118784
[    5.423225] pcpu-alloc: s80504 r8192 d30088 u118784 alloc=29*4096
[    5.423370] Detected VIPT I-cache on CPU0
[    5.930004] EFI services will not be available.
[    5.932370] GICv3: CPU2: using allocated LPI pending table @0x00000001001d0000
[    5.933251] GICv3: CPU3: found redistributor 300 region 0:0x00000000fe6e0000
[    5.933312] CPU3: Booted secondary processor 0x0000000300 [0x412fd050]
[    5.934031] CPU features: detected: Spectre-v4
[    5.934036] CPU features: detected: Spectre-BHB
[    5.934040] Detected PIPT I-cache on CPU4
[    5.934096] GICv3: CPU4: found redistributor 400 region 0:0x00000000fe700000
[    5.937005] CPU features: detected: CRC32 instructions
[    5.937020] CPU features: detected: RCpc load-acquire (LDAPR)
[    5.937033] CPU features: detected: Privileged Access Never
[    5.937047] CPU features: detected: RAS Extension Support
[    5.937062] CPU features: detected: Speculative Store Bypassing Safe (SSBS)
[    5.957730] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6370867519511994 ns
[    5.957744] futex hash table entries: 2048 (order: 5, 131072 bytes, linear)
[    5.960300] Registered FIQ tty driver
[    5.962980] ramoops: pmsg    0x30000@0x00000000001c0000
[    6.015408] rockchip-gpio fec30000.gpio: probed /pinctrl/gpio@fec30000
[    6.534059] reg-fixed-voltage vcc12v-dcin: vcc12v_dcin supplying 12000000uV
[    6.534370] reg-fixed-voltage vcc5v0-usbdcin: vcc5v0_usbdcin supplying 5000000uV
[    6.591305] vcc3v3_sys: 3300 mV, enabled
[    6.591382] reg-fixed-voltage vcc3v3-sys: vcc3v3_sys supplying 3300000uV
[    6.591906] libata version 3.00 loaded.
[    6.610422] TCP established hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    6.612775] TCP: Hash tables configured (established 131072 bind 65536)
[    6.613208] UDP hash table entries: 8192 (order: 7, 786432 bytes, linear)
[    6.648380]    32regs          :  8100 MB/sec
[    6.662864] rk-pcie fe180000.pcie:       IO 0x00f3100000..0x00f31fffff -> 0x00f3100000
[    6.662926] rk-pcie fe180000.pcie:      MEM 0x00f3200000..0x00f3ffffff -> 0x00f3200000
[    6.665725] mpp_rkvdec2 fdc30000.rkvdec-ccu: rkvdec-ccu, probing start
[    6.665799] mpp_rkvdec2 fdc30000.rkvdec-ccu: ccu-mode: 1
[    6.665808] mpp_rkvdec2 fdc30000.rkvdec-ccu: probing finish
[    6.677797] dma-pl330 fea10000.dma-controller: Loaded driver for PL330 DMAC-241330
[    6.677817] dma-pl330 fea10000.dma-controller:       DBUFF-128x8bytes Num_Chans-8 Num_Peri-32 Num_Events-16
[    6.678867] dma-pl330 fea30000.dma-controller: Loaded driver for PL330 DMAC-241330
[    6.678883] dma-pl330 fea30000.dma-controller:       DBUFF-128x8bytes Num_Chans-8 Num_Peri-32 Num_Events-16
[    6.679954] dma-pl330 fed10000.dma-controller: Loaded driver for PL330 DMAC-241330
[    6.679971] dma-pl330 fed10000.dma-controller:       DBUFF-128x8bytes Num_Chans-8 Num_Peri-32 Num_Events-16
[    6.680729] rockchip-pvtm fdb30000.pvtm: pvtm@4 probed
[    6.681320] rockchip-system-monitor rockchip-system-monitor: system monitor probe
[    6.690340] rk_iommu fdca0000.iommu: av1d iommu enabled
[    6.701978] rockchip-vop2 fdd90000.vop: [drm:vop2_bind] vp3 assign plane mask: Cluster3 | Esmart3[0x280], primary plane phy id: Cluster3[7]
[    6.713304] rockchip-vop2 fdd90000.vop: leakage=40
[... 124 lines cut ...]
[    7.296302] usb usb1: SerialNumber: fc840000.usb
[    7.347328] usbcore: registered new interface driver uvcvideo
[    7.456661] usbcore: registered new interface driver usbhid
[    7.456663] usbhid: USB HID core driver
[    7.458812] usbcore: registered new interface driver snd-usb-audio
[    7.796241] usb 1-1: new full-speed USB device number 2 using ohci-platform
[    8.016649] usb 1-1: New USB device found, idVendor=046d, idProduct=c52b, bcdDevice=24.11
[    8.018976] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    8.021243] usb 1-1: Product: USB Receiver
[    8.023475] usb 1-1: Manufacturer: Logitech
[    8.069196] input: Logitech USB Receiver as /devices/platform/fc840000.usb/usb1/1-1/1-1:1.0/0003:046D:C52B.0001/input/input3
[    8.126577] hid-generic 0003:046D:C52B.0001: input,hidraw0: USB HID v1.11 Keyboard [Logitech USB Receiver] on usb-fc840000.usb-1/input0
[    8.135588] input: Logitech USB Receiver Mouse as /devices/platform/fc840000.usb/usb1/1-1/1-1:1.1/0003:046D:C52B.0002/input/input5
[    8.196413] input: Logitech USB Receiver Consumer Control as /devices/platform/fc840000.usb/usb1/1-1/1-1:1.1/0003:046D:C52B.0002/input/input6
[    8.199105] input: Logitech USB Receiver System Control as /devices/platform/fc840000.usb/usb1/1-1/1-1:1.1/0003:046D:C52B.0002/input/input7
[    8.202142] hid-generic 0003:046D:C52B.0002: input,hiddev96,hidraw1: USB HID v1.11 Mouse [Logitech USB Receiver] on usb-fc840000.usb-1/input1
[    8.209608] hid-generic 0003:046D:C52B.0003: hiddev97,hidraw2: USB HID v1.11 Device [Logitech USB Receiver] on usb-fc840000.usb-1/input2

$ lsusb
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 002: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

spele1100, 2025-02-10:
Checked same issue on [ubuntu-22.04-preinstalled-server-arm64-orangepi-cm5.img.xz](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v2.4.0/ubuntu-22.04-preinstalled-server-arm64-orangepi-cm5.img.xz)

Don-Swanson, 2025-03-08:
Same issue here except I'm on the OrangePi 5 Plus

closed 2025-09-08 by github-actions[bot]