Joshua-Riek/ubuntu-rockchip #1200: Bug Report: Problems with Mali drivers – versions mismatch?

tommiv, 2025-03-01:
### What happened?

Trying to make Jellyfin to work with hardware transcoding acceleration, but I have problems with binaries versions mismatch:

```bash
dmesg | grep mali | grep DDK
[   47.314926] mali fb000000.gpu: Kernel DDK version g18p0-01eac0
```
note `g18p0` here.

Jellyfin manual asks to install [this .deb](https://github.com/tsukumijima/libmali-rockchip/releases/download/v1.9-1-2d267b0/libmali-valhall-g610-g13p0-gbm_1.9-1_arm64.deb) in particular, "For the 6.1 LTS kernel on [Ubuntu-Rockchip](https://github.com/Joshua-Riek/ubuntu-rockchip)".
It doesn't work, and I assume it's because of `g13p0` version it's built against.

In [this issue](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/879) there is an advice to just install `ibmali-g610-x11` with `apt`. Sounds cool but doesn't work either.

In both cases, when I run `clinfo --list` I get some errors about missing kernel mod:
```bash
clinfo --list
arm_release_ver: g13p0-01eac0, rk_so_ver: 3
Failed creating base context during opening of kernel driver.
Kernel module may not have been loaded
Platform #0: ARM Platform
arm_release_ver: g13p0-01eac0, rk_so_ver: 3
Failed creating base context during opening of kernel driver.
Kernel module may not have been loaded
 `-- Device #0: <printDeviceInfo:0: get CL_DEVICE_NAME size : error -6>
```
and there are two lines suggesting `g13p0`. 

I'm not so competent in this, but my assumption is to make it work properly, I need both firmware's DDK and libmali DDK to be the same – g13p0, or g18p0. Unfortunately, I can't find a libmali binary for g18p0 anywhere, and I have no idea how to downgrade the blob to g13p0 safely.

Are there any suggestions?

### Kernel version

6.1.0-1025-rockchip

### SBC model

Orange Pi 5

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

### Relevant logs

```shell

```

Not-a-good-username, 2025-04-06:
你好，不知道你是否已经解决，我同样使用RK3588（orangepi 5 plus）运行jellyfin，并且目前硬件解码处于大部分可用的状态。就我个人的经验以及一些帖子的观察来看，userland和kernal的版本差异可能并不会造成不可用的情况，所以你的担心是多余的，你提到的问题我也经常碰到，我的解决方案是，卸载所有与libmali相关的配置后，先运行sudo apt install libmali-g610-x11，随即按照提示安装jellyfin官网指向的libmali驱动（sudo apt install ./libmali-valhall-g610-g13p0-gbm_1.9-1_arm64.deb），重启后即可实现正常运行。

Hi there! I noticed you mentioned an issue, and I wanted to share that I'm also using the RK3588 (Orange Pi 5 Plus) to run Jellyfin, with hardware decoding mostly functional. From my experience and observations in various posts, the differences between userland and kernel versions don't necessarily cause critical issues, so your concerns might be unfounded. I've encountered the problem you described before, and my solution was to uninstall all libmali-related configurations, then run sudo apt install libmali-g610-x11, followed by installing the libmali driver recommended on the Jellyfin website as prompted （sudo apt install ./libmali-valhall-g610-g13p0-gbm_1.9-1_arm64.deb）. After a reboot, everything should work properly.

以下是jellyfin官网提供的验证命令(sudo /usr/lib/jellyfin-ffmpeg/ffmpeg -v debug -init_hw_device rkmpp=rk -init_hw_device opencl=ocl@rk)后输出的结果：

The following is the output result after running the verification command (sudo /usr/lib/jellyfin-ffmpeg/ffmpeg -v debug -init_hw_device rkmpp=rk -init_hw_device opencl=ocl@rk) provided on the Jellyfin official website:  

```
ffmpeg version 7.0.2-Jellyfin Copyright (c) 2000-2024 the FFmpeg developers
  built with gcc 13 (Ubuntu 13.3.0-6ubuntu2~24.04)
  configuration: --prefix=/usr/lib/jellyfin-ffmpeg --target-os=linux --extra-version=Jellyfin --disable-doc --disable-ffplay --disable-ptx-compression --disable-static --disable-libxcb --disable-sdl2 --disable-xlib --enable-lto=auto --enable-gpl --enable-version3 --enable-shared --enable-gmp --enable-gnutls --enable-chromaprint --enable-opencl --enable-libdrm --enable-libxml2 --enable-libass --enable-libfreetype --enable-libfribidi --enable-libfontconfig --enable-libharfbuzz --enable-libbluray --enable-libmp3lame --enable-libopus --enable-libtheora --enable-libvorbis --enable-libopenmpt --enable-libdav1d --enable-libsvtav1 --enable-libwebp --enable-libvpx --enable-libx264 --enable-libx265 --enable-libzvbi --enable-libzimg --enable-libfdk-aac --arch=arm64 --cross-prefix=/usr/bin/aarch64-linux-gnu- --toolchain=hardened --enable-cross-compile --enable-rkmpp --enable-rkrga
  libavutil      59.  8.100 / 59.  8.100
  libavcodec     61.  3.100 / 61.  3.100
  libavformat    61.  1.100 / 61.  1.100
  libavdevice    61.  1.100 / 61.  1.100
  libavfilter    10.  1.100 / 10.  1.100
  libswscale      8.  1.100 /  8.  1.100
  libswresample   5.  1.100 /  5.  1.100
  libpostproc    58.  1.100 / 58.  1.100
Splitting the commandline.
Reading option '-v' ... matched as option 'v' (set logging level) with argument 'debug'.
Reading option '-init_hw_device' ... matched as option 'init_hw_device' (initialise hardware device) with argument 'rkmpp=rk'.
Reading option '-init_hw_device' ... matched as option 'init_hw_device' (initialise hardware device) with argument 'opencl=ocl@rk'.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option v (set logging level) with argument debug.
Applying option init_hw_device (initialise hardware device) with argument rkmpp=rk.
Applying option init_hw_device (initialise hardware device) with argument opencl=ocl@rk.
arm_release_ver: g13p0-01eac0, rk_so_ver: 3
[AVHWDeviceContext @ 0xaaaaff6eaf00] 1 OpenCL platforms found.
[AVHWDeviceContext @ 0xaaaaff6eaf00] 1 OpenCL devices found on platform "ARM Platform".
[AVHWDeviceContext @ 0xaaaaff6eaf00] 0.0: ARM Platform / Mali-G610 r0p0
[AVHWDeviceContext @ 0xaaaaff6eaf00] cl_arm_import_memory found as platform extension.
[AVHWDeviceContext @ 0xaaaaff6eaf00] cl_khr_image2d_from_buffer found as platform extension.
[AVHWDeviceContext @ 0xaaaaff6eaf00] DRM to OpenCL mapping on ARM function found (clImportMemoryARM).
Successfully parsed a group of options.
Universal media converter
usage: ffmpeg [options] [[infile options] -i infile]... {[outfile options] outfile}...

Use -h to get full help or, even better, run 'man ffmpeg'
``` 

但目前困扰我的是，一段时间后或是执行apt更新后，有时会存在驱动失效的情况，需要重新进行上述步骤以实现硬件解码。我并不知道这出于什么原因。此外问题还包括对于杜比视界profile 5没有办法实现色调映射（jellyfin-ffmpeg的主要贡献人nyamisaka提到过这个问题好像只能等待ffmpeg上游解决问题，目前无解）

However, I've been dealing with an issue where, after some time or after running apt upgrade, the driver occasionally stops working, requiring me to repeat the above steps to restore hardware decoding. I don’t know why this is the case. Additionally, there's a problem with Dolby Vision Profile 5 tone mapping, which Jellyfin-FFmpeg contributor nyamisaka mentioned is currently unresolved and will likely need to wait for a fix from the upstream FFmpeg project.

我不知道这是否能帮助你，事实上，由于硬件解码功能的缺失（AV1 编码，杜比视界Profile 5色调映射等）以及相关操作确实过于复杂，Jellyfin在rkmpp上的支持在未来仍然不明朗，我已经在考虑迁移至X86平台，但我仍对即将发布的RK3688抱有期待。

I don't know if this will help you, but in reality, due to the lack of hardware decoding capabilities （AV1  encoding, Dolby Vision Profile 5 tone mapping etc.）and the complexity of the operations involved, the future support for Jellyfin on rkmpp remains uncertain. I have already been considering migrating to the x86 platform, but I still hold some expectations for the upcoming RK3688.

tommiv, 2025-04-06:
@Not-a-good-username thank you for such a detailed answer. Unfortunately, the Jellyfin recommended binary doesn't work for me, because Mali firmware uses g18p0 DDK version on my machine, while the driver uses g13. I don't know how I got g18 and I haven't found an easy way to downgrade this, that's why I created this ticket.

I solved the problem in radical way though, I just bought ASUS NUC with amd64 CPU and installed a conventional Ubuntu Server on it. It's much, much more expensive than any orange pi, but after a few days of struggle I decided that my weekends time is more valuable.

Not-a-good-username, 2025-04-06:
> [@Not-a-good-username](https://github.com/Not-a-good-username) thank you for such a detailed answer. Unfortunately, the Jellyfin recommended binary doesn't work for me, because Mali firmware uses [...]

DDK version doesn't matter, for me DDK version is also g18p0，and g13 is working just as i said before.
And indeed, amd64 save a lot of time！

closed 2025-10-06 by github-actions[bot]