Joshua-Riek/ubuntu-rockchip #1209: Bug Report: PWM Fan connector not working on Orange Pi 5 Max

DiegoBM, 2025-03-11:
### What happened?

Using Image `Ubuntu preinstalled server 24.04 v2.3.2` on orange pi 5 max, the fan connected to the designated 1.25mm pwm fan connector won't do anything at all no matter the cpu load. I tested the same using orange pi's image `Orangepi5max_1.0.0_ubuntu_jammy_server_linux5.10.160` and the fan work under cpu stress.

### Kernel version

6.1.0-1023

### SBC model

Orange Pi 5 Max

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

### Relevant logs

```shell

```

labmaster, 2025-03-13:
I use the Fan connector on a Orange Pi 5 Ultra (beside the different HDMI feature, nearly the same as the Orange Pi 5 Max ) with 
"ubuntu-24.04-preinstalled-server-arm64-orangepi-5-max"  and there the connector works.
The core temp has to rise quite high (above 60°C) before it switches on the fan but in general i works.

DiegoBM, 2025-03-13:
Thank you for your reply @labmaster. It went well above 60 and did hurt to the touch, so I presume there must be something else going on here. Maybe there's any setting I can configure?

DiegoBM, 2025-06-21:
I guess this will go down with the whole project

DiegoBM, 2025-09-22:
Keeping it current