Joshua-Riek/ubuntu-rockchip #1214: Hack for running this on the rock-5t

rafayahmed317, 2025-03-19:
If someone wants to use this distribution on the Radxa Rock 5T, you can use the Radxa Rock 5B Plus image (https://joshua-riek.github.io/ubuntu-rockchip-download/) with the dtb from this [repo](https://github.com/rafayahmed317/Rock-5T-DTB.git) . Just load the image like you normally would and once you get it running, clone the repo and follow the instructions. This works for me to enable the additional Ethernet port, and the Intel Wi-Fi and Bluetooth card on the Rock 5T.

rafayahmed317, 2025-03-19:
Also, this is not a bug or an issue, just thought I would share it here to help others out.

Lemon1151, 2025-04-17:
> If someone wants to use this distribution on the Radxa Rock 5T, you can use the Radxa Rock 5B Plus image (https://joshua-riek.github.io/ubuntu-rockchip-download/) with the dtb from this [...]

https://github.com/Lemon1151/ubuntu-rockchip/releases/download/Ubuntu-2025-04/ubuntu-24.04-noble-desktop-arm64-rock-5t.img.xz    
       You can try this.

closed 2025-10-20 by github-actions[bot]