Joshua-Riek/ubuntu-rockchip #1215: Feature Request: Support smart AM60 RK3588

wishday, 2025-03-24:
### Feature Description

Hope smart AM60 with rk3588 can be supported. 

The Ubuntu system here featuring a desktop environment and powered by the Panfrost GPU driver operates with notable smoothness and stability！

**Product description：**
SoC：RK3588
Memory：8~16GB
Storage：64~256GB
Details：https://downloads.smarttech.com/media/sitecore/en/support/accessory/am60/specs/specsam60.pdf

**DTB from ophub**
https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/build-armbian/armbian-files/platform-files/rockchip/bootfs/dtb/rockchip/rk3588-smart-am60.dtb

**Uboot from ophub**
https://github.com/ophub/u-boot/tree/main/u-boot/rockchip/smart-am60

**Wifi & BT module from ophub**
https://github.com/ophub/firmware/tree/main/firmware/ap6276p

Joshua-Riek, 2025-03-27:
Are you not able to send a pull request?

wishday, 2025-03-28:
**Thank you for your response!**
First, I apologize for taking up your time. I am an ordinary user of the project, with my technical background primarily focused on upper-layer application deployment (such as Docker container management, web service configuration, etc.). I have no experience in low-level technologies like Linux system migration, kernel compilation, or code development.

**I greatly admire the RK3588 GPU Panfrost and rkmm hardware-accelerated desktop solution implemented by the project.** 
I have previously self-learned through Google and attempted to add desktop GPU acceleration functionality based on Ophub's Armbian firmware, but all attempts failed. Here are the details of my trials:

1. Using lsmod, I confirmed that Ophub's Linux 6.1.99 kernel supports Panfrost. After installing mali-g610-firmware from ppa:jjriek/panfork-mesa, the desktop could use the Panfrost driver, but it only worked with X11, not Wayland. Additionally, various apps frequently crashed and were essentially unusable.
2. After compiling and installing Mesa from the source code at https://gitlab.freedesktop.org/mesa/mesa, Panfrost + X11 seemed to activate. The HDMI output showed an X mouse cursor, but I couldn't enter the desktop.
3. After installing the closed-source libmali deb package from https://github.com/tsukumijima/libmali-rockchip and replacing mali_csffw.bin, GPU acceleration still couldn't be activated.

Could your project consider adding support for this device? 
If there is a test version available, I would be willing to assist in verifying the functionality and providing feedback. 

I am fully aware of the precious development resources, and I thank you again for your efforts!

Lemon1151, 2025-04-10:
> Are you not able to send a pull request?

Dear Author,

Hello,I greatly admire the RK3588 GPU Panfrost and rkmm hardware-accelerated desktop solution implemented by your project.

The kernel in your repository is based on the Rockchip 6.1.75 version. Recently, Rockchip updated the kernel to version 6.1.115. I attempted to use the method from your repository to build an Ubuntu 24.04 desktop based on the 6.1.115 kernel. While it can boot into the Ubuntu configuration interface normally, the gnome-shell and Xorg crash continuously during configuration in the setup interface, causing the desktop to restart repeatedly. Even after entering the desktop, the same issue persists. I also tried upgrading the kernel of the previously working Ubuntu 24.04 (based on kernel 6.1.75) to the latest 6.1.115 kernel, but the problem still occurs. Could you please guide me on what might be causing this issue?

wishday, 2025-04-10:
> You need to find the U-Boot configuration file defconfig and the U-Boot device tree dts for AM60. I tried reaching out to the author of the U-Boot submission in the Ophub repository, but I haven't received a response.

**DTS for kernel and uboot are not the same? I think dts file below should be enough.**
https://github.com/armbian/linux-rockchip/blob/rk-6.1-rkr5/arch/arm64/boot/dts/rockchip/rk3588-smart-am60.dts

Lemon1151, 2025-04-10:
> **DTS for kernel and uboot are not the same? I think dts file below should be enough.** https://github.com/armbian/linux-rockchip/blob/rk-6.1-rkr5/arch/arm64/boot/dts/rockchip/rk3588-smart-am60.dts

3588用的UBoot是radxa仓库的2017版本，两个dts不通用

Lemon1151, 2025-04-14:
> **DTS for kernel and uboot are not the same? I think dts file below should be enough.** https://github.com/armbian/linux-rockchip/blob/rk-6.1-rkr5/arch/arm64/boot/dts/rockchip/rk3588-smart-am60.dts

https://github.com/Lemon1151/ubuntu-rockchip/releases/download/Ubuntu-2025-04/ubuntu-24.04-noble-desktop-arm64-smart-am60.img.xz

You can try this firmware. Since there is no defconfig and DTS for the u-boot of this board, it uses the configuration files from another board. There's a high probability that it won't boot up successfully.

Lemon1151, 2025-04-15:
> Dear Author,
>
> Hello,I greatly admire the RK3588 GPU Panfrost and rkmm hardware-accelerated desktop solution implemented by your project.
>
> The kernel in your repository is based on the Rockchip [...]

Hello! I have found the problem and successfully resolved it! It was related to the Mali GPU. I downgraded the Kernel DDK version to g18p0-01eac0, and after building Ubuntu 24.04 based on the Rockchip 6.1.115 kernel, everything is now working fine.

stimandrew, 2025-10-29:
> > > Вы не можете отправить пул-реквест?
> > 
> > 
> > Уважаемый Автор,
> > Здравствуйте! Я восхищаюсь графическим процессором RK3588 Panfrost и решением для настольных компьютеров с аппаратным ускорением rkmm, реализованным в вашем проекте.
> > Ядро в вашем репозитории основано на версии Rockchip 6.1.75. Недавно компания Rockchip обновила ядро до версии 6.1.115. Я попытался использовать метод из вашего репозитория для создания рабочего стола Ubuntu 24.04 на основе ядра 6.1.115. Хотя система может нормально загружаться в интерфейсе настройки Ubuntu, gnome-shell и Xorg постоянно дают сбой во время настройки в интерфейсе установки, что приводит к многократному перезапуску рабочего стола. Даже после входа в систему проблема сохраняется. Я также попытался обновить ядро ранее работавшей Ubuntu 24.04 (на основе ядра 6.1.75) до последней версии 6.1.115, но проблема осталась. Не могли бы вы подсказать, в чём может быть причина?
> 
> Здравствуйте! Я нашёл проблему и успешно её решил! Она была связана с графическим процессором Mali. Я понизил версию Kernel DDK до g18p0-01eac0, и после сборки Ubuntu 24.04 на основе ядра Rockchip 6.1.115 всё заработало как надо.

Panfrost заработал на Wayland?