Joshua-Riek/ubuntu-rockchip #1220: Bug Report: Orange Pi 5 Pro does not detect OV13855

hao004, 2025-04-01:
### What happened?

Hi. I downloaded Joshua's Ubuntu 24.04 image and ran sudo apt-get update followed by sudo apt-get install camera-engine-rkaiq-rk3588 but OV13855 is not detected. I tried to look at v4l2-ctl --list-devices but it says 'Cannot open device /dev/video0'. Running ls /dev/video* outputs only /dev/video-dec0 and /dev/video-enc0 only.

I also tried with Joshua's 22.04 ubuntu image and ubuntu images from Orange Pi's website by following the steps given here https://drive.google.com/drive/folders/1j3gmf31XBuKPBeNIQOqqh9X_7SFCOv0s but the same issue occurred. Camera connections to the board are correct. Could it be drivers' problems? Thanks!

### Kernel version

6.1

### SBC model

Orange Pi 5 Pro

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

### Relevant logs

```shell

```

ScheglyakStas, 2025-11-22:
I have the same issue on my Orange Pi 5 Pro. Has anyone found a fix?

katafoxi, 2025-12-03:
Hello.

If the cameras are physically connected correctly (it's possible to make a mistake here), we need to check if they are enabled in software.
```bash
cat /boot/extlinux/extlinux.conf
```
Original state:
```txt
 label l0
         menu label Ubuntu 24.04.1 LTS 6.1.0-1025-rockchip
         linux /boot/vmlinuz-6.1.0-1025-rockchip
         initrd /boot/initrd.img-6.1.0-1025-rockchip
         fdt /lib/firmware/6.1.0-1025-rockchip/device-tree/rockchip/rk3588s-orangepi-5b.dtb
         append root=UUID=10785ddd-97a8-4ae7-a149-67fb848ff8a2 rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory
```
If the `fdtoverlays` line is missing, it means the cameras are simply not enabled in software.

To enable the cameras, you need to apply the corresponding device tree overlays to the corresponding physical MIPI ports.
According to the Orange Pi 5B documentation, two MIPI cameras can work simultaneously on ports C1 and C3.

Let's search for existing overlays for the specific cameras (if any).
```bash
sudo find / -type f -name "*ov1385*.dtb*" 2>/dev/null
```
OUTPUT
```txt
 /usr/lib/firmware/6.1.0-1025-rockchip/device-tree/rockchip/overlay/orangepi-5-plus-ov13855.dtbo
 /usr/lib/firmware/6.1.0-1025-rockchip/device-tree/rockchip/overlay/orangepi-5-plus-ov13850.dtbo
 /usr/lib/firmware/6.1.0-1025-rockchip/device-tree/rockchip/overlay/orangepi-5-ov13855-c1.dtbo
 /usr/lib/firmware/6.1.0-1025-rockchip/device-tree/rockchip/overlay/orangepi-5-ov13855-c2.dtbo
 /usr/lib/firmware/6.1.0-1025-rockchip/device-tree/rockchip/overlay/orangepi-5-ov13850-c3.dtbo
 /usr/lib/firmware/6.1.0-1025-rockchip/device-tree/rockchip/overlay/orangepi-5-ov13850-c2.dtbo
 /usr/lib/firmware/6.1.0-1025-rockchip/device-tree/rockchip/overlay/orangepi-5-ov13850-c1.dtbo
 /usr/lib/firmware/6.1.0-1025-rockchip/device-tree/rockchip/overlay/orangepi-5-ov13855-c3.dtbo
```
So, the available overlays can be found in `/lib/firmware/6.1.0-1025-rockchip/device-tree/rockchip/overlay`. Choose the ones you need and add them to `/etc/default/u-boot`.
Note: OV13850 and OV13855 are different camera models, so make sure to specify which camera is on which port.

```bash
sudo nano /etc/default/u-boot
```
To enable OV13850 and OV13855 cameras, uncomment and modify the `U_BOOT_FDT_OVERLAYS` and `U_BOOT_FDT_OVERLAYS_DIR` lines.

```txt
U_BOOT_FDT_OVERLAYS="orangepi-5-ov13850-c1.dtbo orangepi-5-ov13855-c3.dtbo"
U_BOOT_FDT_OVERLAYS_DIR="/lib/firmware/6.1.0-1025-rockchip/device-tree/rockchip/overlay"
```
After saving `/etc/default/u-boot` file:
```bash
sudo u-boot-update
sudo reboot
```

After rebooting, check again:
```bash
cat /boot/extlinux/extlinux.conf
```
New state:
```txt
 label l0
         menu label Ubuntu 24.04.1 LTS 6.1.0-1025-rockchip
         linux /boot/vmlinuz-6.1.0-1025-rockchip
         initrd /boot/initrd.img-6.1.0-1025-rockchip
         fdt /lib/firmware/6.1.0-1025-rockchip/device-tree/rockchip/rk3588s-orangepi-5b.dtb
         fdtoverlays  /lib/firmware/6.1.0-1025-rockchip/device-tree/rockchip/overlay/orangepi-5-ov13850-c1.dtbo /lib/firmware/6.1.0-1025-rockchip/device-tree/rockchip/overlay/orangepi-5-ov13855-c3.dtbo
         append root=UUID=10785ddd-97a8-4ae7-a149-67fb848ff8a2 rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory
v4l2-ctl --list-devices
```
All the best!