Joshua-Riek/ubuntu-rockchip #1229: Bug Report: Can't boot on Orange pi 5 plus

alexxony, 2025-04-27:
### What happened?

I need to run rknn models but it needs higher rknpu driver which is 0.98.

[image]

both images from orange pi official . And their rknpu driver versions are lower than 0.98

[image]
So I downloaded this image
https://joshua-riek.github.io/ubuntu-rockchip-download/boards/orangepi-5-plus.html
 ,extracted  and wrote with balena etcher. 

I tried to boot twice, it did not work. 

only red light turned on board.

How can I solve this problem?

### Kernel version

5.10

### SBC model

Orange pi 5 plus

### What operating system are you seeing this problem on?

Ubuntu 24.04 LTS (Noble Nombat)

### Relevant logs

```shell

```

Lemon1151, 2025-05-03:
> ### What happened?
> I need to run rknn models but it needs higher rknpu driver which is 0.98.
>
> [image]
>
> both images from orange pi official . And their rknpu driver versions are lower than 0.98 [...]

The version of rknpu in the kernel of this repository is still 0.97. You can try whether this firmware works.
https://github.com/Lemon1151/ubuntu-rockchip/releases/download/Ubuntu-2025-04/ubuntu-24.04-noble-desktop-arm64-orangepi-5-plus.img.xz

EatPrilosec, 2025-05-03:
`ubuntu-24.10-preinstalled-desktop-arm64-orangepi-5-plus.img.xz` boots fine for me on my Opi5Plus 16GB

also there are some nightlys of plucky (25.04) in the actions section, idk if they even work though.

https://github.com/Joshua-Riek/ubuntu-rockchip/actions/workflows/nightly.yml

[Nightly Build #346:]
(https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/14805319049) [ubuntu-25.04-preinstalled-desktop-arm64-orangepi-5-plus.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/14805319049/artifacts/3054606365)

Lemon1151, 2025-05-04:
> ubuntu-24.10-preinstalled-desktop-arm64-orangepi-5-plus.img.xz   It should be using the 6.11 kernel, and it probably doesn't include the rknpu driver.

alexxony, 2025-05-11:
> `ubuntu-24.10-preinstalled-desktop-arm64-orangepi-5-plus.img.xz` boots fine for me on my Opi5Plus 16GB
> 
> also there are some nightlys of plucky (25.04) in the actions section, ~idk if they even work though.~ (it boots to desktop at least)
> 
> https://github.com/Joshua-Riek/ubuntu-rockchip/actions/workflows/nightly.yml
> 
> [Nightly Build #346:](https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/14805319049) [ubuntu-25.04-preinstalled-desktop-arm64-orangepi-5-plus.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/14805319049/artifacts/3054606365)

but it does not match rknn models.  they want 0.98

alexxony, 2025-05-11:
> Has anyone tried the 25.04 on Orange PI 5B? I've burnt it to my SDcard but so far it just refuses to boot and just loads the eMMC OS instead (an older 24.04 beta)

for what reason?? 25.04