Joshua-Riek/ubuntu-rockchip #1235: rk-pcie fe170000.pcie: failed to initialize host

wriede, 2025-05-09:
I have to prefix this by saying I don't know what to make of it, because if it wasn't for the error messages I'd say it's working fine.

I put your ubuntu 24.04 image on an ssd using balena etcher and got it to boot and expand on my Orange Pi 5+ 16GB:

wriede@Athena:~/Downloads$ ls -l ubu*
-rw-r--r--. 1 wriede wriede 4543479808 Apr 28 09:07 ubuntu-24.04-preinstalled-server-arm64-orangepi-5-plus.img
-rw-r--r--. 1 wriede wriede        128 Apr 28 09:12 ubuntu-24.04-preinstalled-server-arm64-orangepi-5-plus.img.xz.sha256

Which then identifies running on the Pi as:

wriede@ubuntu:~$ uname -a
Linux ubuntu 6.1.0-1025-rockchip #25-Ubuntu SMP Mon Aug 26 23:01:14 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux

wriede@ubuntu:~$ cat /etc/debian_version 
trixie/sid

The pcie devices are there, note the NVMe ssd:

wriede@ubuntu:~$ lspci
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
0000:01:00.0 Non-Volatile memory controller: SK hynix Gold P31/BC711/PC711 NVMe Solid State Drive
0003:30:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
0003:31:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0004:40:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
0004:41:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)

But dmesg reports a ton of issues, the last bunch of which make it to the console immediately after the login prompt (It may be relevant that only one of the ethernet ports is connected at this time):

wriede@ubuntu:~$ dmesg | grep pcie
[    5.989902] vcc3v3_pcie30: 3300 mV, disabled
[    5.989951] reg-fixed-voltage vcc3v3-pcie30: Looking up vin-supply from device tree
[    5.989957] vcc3v3_pcie30: supplied by vcc5v0_sys
[    5.989988] reg-fixed-voltage vcc3v3-pcie30: vcc3v3_pcie30 supplying 3300000uV
[    5.990236] vcc3v3_pcie2x1l0: 3300 mV, disabled
[    5.990284] reg-fixed-voltage vcc3v3-pcie2x1l0: Looking up vin-supply from device tree
[    5.990289] vcc3v3_pcie2x1l0: supplied by vcc12v_dcin
[    5.990324] reg-fixed-voltage vcc3v3-pcie2x1l0: vcc3v3_pcie2x1l0 supplying 3300000uV
[    5.990382] reg-fixed-voltage vcc3v3-pcie-eth: Looking up vin-supply from device tree
[    5.990388] vcc3v3_pcie_eth: supplied by vcc12v_dcin
[    6.045782] vcc3v3_pcie_eth: 3300 mV, enabled
[    6.045860] reg-fixed-voltage vcc3v3-pcie-eth: vcc3v3_pcie_eth supplying 3300000uV
[    6.117290] dw-pcie fe180000.pcie: invalid resource
[    6.117302] dw-pcie fe180000.pcie: Failed to initialize host
[    6.117311] dw-pcie: probe of fe180000.pcie failed with error -22
[    6.117386] dw-pcie fe190000.pcie: invalid resource
[    6.117395] dw-pcie fe190000.pcie: Failed to initialize host
[    6.117402] dw-pcie: probe of fe190000.pcie failed with error -22
[    6.117520] dw-pcie fe150000.pcie: invalid resource
[    6.117529] dw-pcie fe150000.pcie: Failed to initialize host
[    6.117550] dw-pcie: probe of fe150000.pcie failed with error -22
[    6.117568] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up pcie-supply from device tree
[    6.117590] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up pcie-supply property in node /power-management@fd8d8000/power-controller failed
[    6.117675] dw-pcie fe170000.pcie: invalid resource
[    6.117685] dw-pcie fe170000.pcie: Failed to initialize host
[    6.117692] dw-pcie: probe of fe170000.pcie failed with error -22
[    6.118376] rk-pcie fe180000.pcie: invalid prsnt-gpios property in node
[    6.118410] rk-pcie fe180000.pcie: Looking up vpcie3v3-supply from device tree
[    6.118424] rk-pcie fe180000.pcie: Looking up vpcie3v3-supply property in node /pcie@fe180000 failed
[    6.118437] rk-pcie fe190000.pcie: invalid prsnt-gpios property in node
[    6.118452] rk-pcie fe180000.pcie: no vpcie3v3 regulator found
[    6.118486] rk-pcie fe190000.pcie: Looking up vpcie3v3-supply from device tree
[    6.118499] rk-pcie fe190000.pcie: Looking up vpcie3v3-supply property in node /pcie@fe190000 failed
[    6.118525] rk-pcie fe190000.pcie: no vpcie3v3 regulator found
[    6.118601] rk-pcie fe150000.pcie: invalid prsnt-gpios property in node
[    6.118616] rk-pcie fe150000.pcie: Looking up vpcie3v3-supply from device tree
[    6.118805] rk-pcie fe170000.pcie: invalid prsnt-gpios property in node
[    6.118822] rk-pcie fe170000.pcie: Looking up vpcie3v3-supply from device tree
[    6.119029] rk-pcie fe180000.pcie: host bridge /pcie@fe180000 ranges:
[    6.119077] rk-pcie fe190000.pcie: host bridge /pcie@fe190000 ranges:
[    6.119083] rk-pcie fe180000.pcie:       IO 0x00f3100000..0x00f31fffff -> 0x00f3100000
[... 50 lines cut ...]
[    6.449927] reg-fixed-voltage pcie30-avdd1v8: Looking up vin-supply from device tree
[    6.449936] pcie30_avdd1v8: supplied by avcc_1v8_s0
[    6.450120] pcie30_avdd1v8: 1800 mV, enabled
[    6.450171] reg-fixed-voltage pcie30-avdd1v8: pcie30_avdd1v8 supplying 1800000uV
[    6.458881] pcieport 0000:00:00.0: PME: Signaling with IRQ 159
[    6.467918] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[    6.497935] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[    6.524659] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[    6.551338] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[    9.111274] rk-pcie fe170000.pcie: PCIe Link Fail, LTSSM is 0x3, hw_retries=0
[   11.337986] rk_pcie_establish_link: 127 callbacks suppressed
[   11.337993] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[   11.364590] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[   11.391378] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[   11.417924] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[   11.444652] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[   11.474710] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[   11.504722] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[   11.531262] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[   11.557983] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[   11.584612] rk-pcie fe170000.pcie: PCIe Linking... LTSSM is 0x3
[   13.121262] rk-pcie fe170000.pcie: PCIe Link Fail, LTSSM is 0x3, hw_retries=1
[   14.197986] rk-pcie fe170000.pcie: failed to initialize host

Since the word "power" appears several times I checked whether my power adapter was marginal (it's a 4A/20W one) but that is not the case. I put a tester on the USB-C line and I see the supply at 5.137V and the Pi taking about 1.05 Amp (fluctuates a little). Looks just fine. It's idle so low wattage is expected, but booting isn't exactly a heavy load either.

DiegoBM, 2025-09-19:
Having this same issue as well. Not sure if it happens after the unnatended-upgrades or before. I have reflashed so many time I've lost track

glardies-clintell, 2025-11-02:
Same here (OrangePi 5 Plus):

```shell
ubuntu@ubuntu:~$ dmesg | grep -i pcie
[    6.149103] vcc3v3_pcie30: 3300 mV, disabled
[    6.149154] reg-fixed-voltage vcc3v3-pcie30: Looking up vin-supply from device tree
[    6.149159] vcc3v3_pcie30: supplied by vcc5v0_sys
[    6.149190] reg-fixed-voltage vcc3v3-pcie30: vcc3v3_pcie30 supplying 3300000uV
[    6.149450] vcc3v3_pcie2x1l0: 3300 mV, disabled
[    6.149500] reg-fixed-voltage vcc3v3-pcie2x1l0: Looking up vin-supply from device tree
[    6.149506] vcc3v3_pcie2x1l0: supplied by vcc12v_dcin
[    6.149541] reg-fixed-voltage vcc3v3-pcie2x1l0: vcc3v3_pcie2x1l0 supplying 3300000uV
[    6.149603] reg-fixed-voltage vcc3v3-pcie-eth: Looking up vin-supply from device tree
[    6.149608] vcc3v3_pcie_eth: supplied by vcc12v_dcin
[    6.205624] vcc3v3_pcie_eth: 3300 mV, enabled
[    6.205701] reg-fixed-voltage vcc3v3-pcie-eth: vcc3v3_pcie_eth supplying 3300000uV
[    6.279747] dw-pcie fe180000.pcie: invalid resource
[    6.279761] dw-pcie fe180000.pcie: Failed to initialize host
[    6.279769] dw-pcie: probe of fe180000.pcie failed with error -22
[    6.279846] dw-pcie fe190000.pcie: invalid resource
[    6.279855] dw-pcie fe190000.pcie: Failed to initialize host
[    6.279863] dw-pcie: probe of fe190000.pcie failed with error -22
[    6.279986] dw-pcie fe150000.pcie: invalid resource
[    6.279995] dw-pcie fe150000.pcie: Failed to initialize host
[    6.280010] dw-pcie: probe of fe150000.pcie failed with error -22
[    6.280082] dw-pcie fe170000.pcie: invalid resource
[    6.280091] dw-pcie fe170000.pcie: Failed to initialize host
[    6.280099] dw-pcie: probe of fe170000.pcie failed with error -22
[    6.280396] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up pcie-supply from device tree
[    6.280416] rockchip-pm-domain fd8d8000.power-management:power-controller: Looking up pcie-supply property in node /power-management@fd8d8000/power-controller failed
[    6.280786] rk-pcie fe180000.pcie: invalid prsnt-gpios property in node
[    6.280822] rk-pcie fe180000.pcie: Looking up vpcie3v3-supply from device tree
[    6.280838] rk-pcie fe190000.pcie: invalid prsnt-gpios property in node
[    6.280838] rk-pcie fe180000.pcie: Looking up vpcie3v3-supply property in node /pcie@fe180000 failed
[    6.280863] rk-pcie fe190000.pcie: Looking up vpcie3v3-supply from device tree
[    6.280874] rk-pcie fe190000.pcie: Looking up vpcie3v3-supply property in node /pcie@fe190000 failed
[    6.280888] rk-pcie fe180000.pcie: no vpcie3v3 regulator found
[    6.280902] rk-pcie fe190000.pcie: no vpcie3v3 regulator found
[    6.281036] rk-pcie fe150000.pcie: invalid prsnt-gpios property in node
[    6.281050] rk-pcie fe150000.pcie: Looking up vpcie3v3-supply from device tree
[    6.281389] rk-pcie fe170000.pcie: invalid prsnt-gpios property in node
[    6.281424] rk-pcie fe170000.pcie: Looking up vpcie3v3-supply from device tree
[    6.281491] rk-pcie fe180000.pcie: host bridge /pcie@fe180000 ranges:
[    6.281526] rk-pcie fe180000.pcie:       IO 0x00f3100000..0x00f31fffff -> 0x00f3100000
[    6.281575] rk-pcie fe180000.pcie:      MEM 0x00f3200000..0x00f3ffffff -> 0x00f3200000
[    6.281647] rk-pcie fe180000.pcie:      MEM 0x09c0000000..0x09ffffffff -> 0x09c0000000
[    6.281727] rk-pcie fe180000.pcie: iATU unroll: enabled
[    6.281726] rk-pcie fe190000.pcie: host bridge /pcie@fe190000 ranges:
[    6.281741] rk-pcie fe190000.pcie:       IO 0x00f4100000..0x00f41fffff -> 0x00f4100000
[    6.281747] rk-pcie fe180000.pcie: iATU regions: 8 ob, 8 ib, align 64K, limit 8G
[    6.281763] rk-pcie fe190000.pcie:      MEM 0x00f4200000..0x00f4ffffff -> 0x00f4200000
[    6.281791] rk-pcie fe190000.pcie:      MEM 0x0a00000000..0x0a3fffffff -> 0x0a00000000
[    6.281841] rk-pcie fe190000.pcie: iATU unroll: enabled
[    6.281850] rk-pcie fe190000.pcie: iATU regions: 8 ob, 8 ib, align 64K, limit 8G
[    6.297576] rk-pcie fe150000.pcie: host bridge /pcie@fe150000 ranges:
[    6.297613] rk-pcie fe150000.pcie:       IO 0x00f0100000..0x00f01fffff -> 0x00f0100000
[    6.297640] rk-pcie fe150000.pcie:      MEM 0x00f0200000..0x00f0ffffff -> 0x00f0200000
[    6.297657] rk-pcie fe150000.pcie:      MEM 0x0900000000..0x093fffffff -> 0x0900000000
[    6.297718] rk-pcie fe150000.pcie: iATU unroll: enabled
[    6.297729] rk-pcie fe150000.pcie: iATU regions: 8 ob, 8 ib, align 64K, limit 8G
[    6.341852] rk-pcie fe170000.pcie: host bridge /pcie@fe170000 ranges:
[    6.341919] rk-pcie fe170000.pcie:       IO 0x00f2100000..0x00f21fffff -> 0x00f2100000
[    6.341973] rk-pcie fe170000.pcie:      MEM 0x00f2200000..0x00f2ffffff -> 0x00f2200000
[    6.342018] rk-pcie fe170000.pcie:      MEM 0x0980000000..0x09bfffffff -> 0x0980000000
[    6.342114] rk-pcie fe170000.pcie: iATU unroll: enabled
[    6.342143] rk-pcie fe170000.pcie: iATU regions: 8 ob, 8 ib, align 64K, limit 8G
[    6.488917] rk-pcie fe190000.pcie: PCIe Linking... LTSSM is 0x3
[    6.488956] rk-pcie fe180000.pcie: PCIe Linking... LTSSM is 0x3
[    6.505692] rk-pcie fe150000.pcie: PCIe Linking... LTSSM is 0x3
[    6.531253] rk-pcie fe150000.pcie: PCIe Linking... LTSSM is 0x2
[... 23 lines cut ...]
[    6.642081] reg-fixed-voltage pcie20-avdd1v8: pcie20_avdd1v8 supplying 1800000uV
[    6.642193] reg-fixed-voltage pcie30-avdd0v75: Looking up vin-supply from device tree
[    6.642253] pcie30_avdd0v75: supplied by avdd_0v75_s0
[    6.642382] pcie30_avdd0v75: 750 mV, enabled
[... cut ...]
[   11.607932] rk-pcie fe150000.pcie: PCIe Linking... LTSSM is 0x2
[   11.634574] rk-pcie fe150000.pcie: PCIe Linking... LTSSM is 0x200003
[   11.664653] rk-pcie fe150000.pcie: PCIe Linking... LTSSM is 0x2
[   11.694641] rk-pcie fe150000.pcie: PCIe Linking... LTSSM is 0x200003
[   11.721308] rk-pcie fe150000.pcie: PCIe Linking... LTSSM is 0x2
[   11.751247] rk-pcie fe150000.pcie: PCIe Linking... LTSSM is 0x3
[   13.137864] rk-pcie fe150000.pcie: PCIe Link Fail, LTSSM is 0x2, hw_retries=1
[   13.137937] rk-pcie fe150000.pcie: failed to initialize host´´´