Joshua-Riek/ubuntu-rockchip #1258: Feature Request: Support for 定昌 DC-A588 (RK3588-based) Board

AndroidOL, 2025-06-08:
# 🧩 Request: Support for DC-A588 (RK3588-based) Board

Hi,

I’m writing to kindly request support for my device, the **DC-A588**, which is based on the **Rockchip RK3588**. I’ve extracted the device tree from the official Ubuntu 20.04 image as well as from the board itself, and uploaded everything here:

- 📁 GitHub Archive: [AndroidOL/Archive-DC-A588](https://github.com/AndroidOL/Archive-DC-A588)
- 📄 Device Tree Blob: [01_dtbdump_rockchip,rk3588-evb7-lp4-v10.dtb](https://github.com/AndroidOL/Archive-DC-A588/blob/main/dtb-from-official-image/01_dtbdump_rockchip%2Crk3588-evb7-lp4-v10.dtb)
- 📚 Official documentation (Chinese): [DC-A588 Wiki](http://wikicn.gzdcsmt.com/wendang_id_8.html#son36)

---

## ✅ Current Status

I’ve tested [**Lemon1151's fork of this project**](https://github.com/Lemon1151/ubuntu-rockchip/commit/e2e9eeb479cf38a828f39cc5eaffbaf35a538b4e), which works fairly well with **kernel 6.1.x**. It supports:

- My display at **2560x1080** resolution (native is 3440x1440; EDID supports 256-byte)
- Better graphics than the official Debian/Ubuntu images, which are limited to **1920x1080** with software rendering only

---

## ⚠️ Remaining Issues

1. **EC20 4G module (M.2 slot):**
   - Not recognized (`lsusb` shows nothing)
   - Kernel is compiled with the `option` driver enabled

2. **TF Card Reader:**
   - Not recognized at all
   - Only `mmc0` is available — no additional `mmc` hosts visible

---

## 🙏 Request

Would it be possible to help bring **mainline/upstream support** for this board (or board variant)?  
The Lemon1151 fork demonstrates that most of the hardware is working. I'm happy to provide more data, run tests, and help in any way possible to assist the integration.

Thank you very much for your excellent work and support!

Best regards,  
**@AndroidOL**

AndroidOL, 2025-06-08:
@Lemon1151 Really hoping someone can help me out.

zhuangguangkang0013, 2025-06-30:
> [@Lemon1151](https://github.com/Lemon1151) Really hoping someone can help me out.

Bro, do you have the source code of DC-A588 Android system? I am looking for it. Thanks