Joshua-Riek/ubuntu-rockchip #1262: Bug Report: Cannot connect to serial console of Orange Pi 5 RK3588S

kristophercrawford, 2025-06-20:
### What happened?

I am attempting to access the Orange Pi 5 over serial. When I connect a usb to serial adapter to the three pins near the power/activity led's, I see no output on the console. 

Thinking that my adapter could not communicate at the set speed, I changed `/boot/firmware/ubuntuEnv.txt` to be `console=ttyS2,115200`. I am still not seeing any output on the console. 

Do I need to change other settings to enable console output on the Orange Pi? 

### Kernel version

5.10.160-rockchip

### SBC model

Orange Pi 5

### What operating system are you seeing this problem on?

Ubuntu 22.04 LTS (Jammy Jellyfish)

### Relevant logs

```shell

```

kristophercrawford, 2025-06-20:
I was using the Ubuntu 22.04 image I downloaded from from here `https://joshua-riek.github.io/ubuntu-rockchip-download/boards/orangepi-5.html`

I have since re-installed using the 24.04 image and still cannot see any output on the serial console.

Joshua-Riek, 2025-06-21:
The path /boot/firmware/ubuntuEnv.txt does not exist.