Joshua-Riek/ubuntu-rockchip #1304: Feature Request: Orange Pi 5 Ultra

moo-the-cow, 2025-09-28:
### Feature Description

There is no armbian version for the orange pi 5 ultra. and the stock noble image provided by rockchip is kind of bad, for example the u-boot SPI flash does not really work to boot into emmc or nvme and of course they only released a "it kinda" works version and stopped.

some infos on that:
http://www.orangepi.org/orangepiwiki/index.php/Orange_Pi_5_Ultra
http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-Pi-5-Ultra.html

moo-the-cow, 2025-09-29:
yeah I saw the other locked thread. people act like it's something trivial, so this armbian custom "build" does not even boot.
I don't want to use another boards image, because there are differences in board hardware layouts.
It will be sure a challenge, which is why I suspect there nothing yet on armbian side

moo-the-cow, 2025-09-29:
the current one (xunlong orangepi) is on jellyfin. it's rather outdated, the only feature I miss though is the nvme and emmc, could be simply a u-boot issue. and hardware encoding/decoding will be important to me (server minimal is fine)

moo-the-cow, 2025-10-17:
no the same actually applies to nvme, which btw I also tried.
I also tried to create a custom orange pi 5 ultra build based on the joshua riek build scripts, but I failed, because simply I lack some of the necessary knowledge

TrevTrev84, 2025-11-01:
Dietpi uses a Bootloader file stored on emmc to boot nvme... maybe this is the way